arkypita/LaserGRBL #1181: Laser On all the time

jmdang17, 2020-11-29:
Hi, I have a problem with my laser, it's keep turning on all the time, how can I fix it?
I'm using CNC Shield, 5.5w laser. I have connected the TTL(red wire) to the Z+ on the board and TTL(black wire) to the pin beside the Z+. Is this correct?
And also, I have another issue where once I connected any of the end stop limit on the board, the alarm will comes out and I can't do anything for it to run. If I disconnected all of the end stop limit switch from the cnc shield, everything is ok.
Attached my connections and codes. Appreciate for the help.
[image]
[image]

[image]

arkypita, 2020-11-29:
Your wiring is ok. Take some time to test TTL feature manually. Insert TTL- to any black pin, and with TTL+  touch the +5v of the cnc shield (yellow pin). Laser should turn on.

Touch the GND pin (blue pin or any black pin) the laser should turn off.

Let me know

About limit switch: one issue at time!

jmdang17, 2020-11-29:
Hi, thanks for the reply. I'm a little confused here. Do you meann the TTL- should insert to any black pin and TTL+ should connect to yellow(+5v) pin? I thought the TTL+ need insert to the Z+, sorry I'm newbie here. Ok, I will test it out. :)

gmmanonymus111, 2020-11-29:
TTL+ indeed should be connected to Z+. What arkypita meant is you need to test your laser drivers functionality. With 5V signal, the laser should turn on (this is why you need to touch the TLL+ to the 5V output of the shield), and when there is no voltage, or it is connected to any of the ground pins, the laser should be turned off. Any value between 5V and 0V means the "intensity" of the laser power. That's what Z+ should do, set the correct voltage for your laser to burn on higher power, or just slightly mark the wood with lower power, depending on your GCode.

arkypita, 2020-11-29:
Ps: wear protective goggles and point laser far from you and flamable object when doing test

arkypita, 2020-11-29:
> What arkypita meant is you need to test your laser drivers functionality. 

Yes, I'm suggesting a diagnostic procedure.

> Take some time to test TTL feature manually. Insert TTL- to any black pin, and with TTL+ touch the +5v of the cnc shield (yellow pin). Laser should turn on.

You must first of all try to understand if your driver is working or not. The TTL pin, driver side, controls the switching on and off of the laser.

If connected to 0V (gnd) the laser must switch off, if connected to 5V the laser must switch on. If it doesn't, your driver won't work properly.

>  Any value between 5V and 0V means the "intensity" of the laser power. That's what Z+ should do, set the correct voltage for your laser to burn on higher power, or just slightly mark the wood with lower power, depending on your GCode.

@gmmanonymus111 
What you say is incorrect. TTL is a digital binary pin. It only recognize 0-1 on/off.

[image]

Power modulation is realized by switching on/off multiple time per second (about 1KHz) and varying duty cycle with a technique called PWM.

[image]

StuartB4, 2020-11-29:
If you have the laser and fan plugs the wrong way the laser would be on all the time becaue the fan connector is a constant voltage to the fan.

jmdang17, 2020-11-30:
> Yes, I'm suggesting a diagnostic procedure.
>
>
> You must first of all try to understand if your driver is working or not. The TTL pin, driver side, controls the switching on and off of the laser.
>
> If [...]

Hi arkypita, I have tested the laser, when I connect the TTL+ to the +5v of the cnc shield (yellow pin) and TTL- to any black pin, the laser is on. And also when I connect the TTL+ to the black pin, only the fan is on, laser is off. So does this means the laser driver is working?
What's the next step should I do?
Thanks much for the reply.

arkypita, 2020-11-30:
> So does this means the laser driver is working?

Yes. Your driver is working.

> What's the next step should I do?

Now you should test arduino & CNC shield side. Do you have a multimeter? Are you able to measure a DC voltage?
Just a question: what version of grbl did you flash on your arduino?

jmdang17, 2020-11-30:
I can borrow a multimeter, how can I test the arduino and CNC shield. Sorry I'm not very good in electrical and wiring stuffs.
I have the version 1.1 for the grbl.

arkypita, 2020-11-30:
As I described, the power is regulated by giving commands from LaserGRBL, which the board receives and processes, which act on pin Z+

In LaserGRBL you have a "type gcode here" box (3)

[image]

Disconnect the laser driver, take your multimeter, put it in Voltage (DC) measurement.
Connect the black probe to a GND pin (i.e any black pin) and left it connected.

With the red probe touch the 5V pin, you should read 5V on multimeter (so you are sure it is set up correctly).

Now connect the red probe to Z+ and left it connected.

In LaserGRBL connect to the engraver and manually send this commands (type a line and hit enter):

$32=0 
M3 G1 F100 S1000

This pair of command should turn on the laser at the max power, so your multimeter should read 5V

type and send 

M5 S0

This command should turn off the laser, so your multimeter should read 0V

If you don't read 0V maybe your arduino is broken.

jmdang17, 2020-12-01:
Hey thanks for the reply. I will get the multimeter and test it out. Just a noob question, the TTL connection from the laser driver will be disconnected while I’m doing the testing process, am I correct?

arkypita, 2020-12-01:
Yes, it is better.

jmdang17, 2020-12-01:
I have just tested the arduino, before I connect the laser driver, I use the red probe touch the 5V pin and black probe touch the black pin, it's 4.66V on multimeter. But when I connect the red probe to Z+ with both the commands below:
1)
M3 S1000
G1 F1000

2)
M5 S0
G0

The laser is not turning off, it's constantly on and the multimeter is showing 4.66V.
So the arduino board can't be use?

jmdang17, 2020-12-01:
Actually could it be my connections? Attached the connections that I have right now, motor is working fine, everything is ok., except the laser constantly on all the time, can help me to check out on this? 
[image]
thanks.

arkypita, 2020-12-01:
Connection Is fine.

> The laser is not turning off, it's constantly on and the multimeter is showing 4.66V.
> So the arduino board can't be use?

There is a final test you can do. Test the arduino board by itself, not connected to the shield, not connected to any power source (just only connected to PC via USB cable).

Test with multimeter (black on GND, red on D11).

[image]

https://github.com/gnea/grbl/wiki/Connecting-Grbl

If you have 4.66v on D11 pin, also when you send M5, your arduino is defective.

arkypita, 2020-12-01:
Pin D11 (the one near Variable spindle PWM) is the one that is connected to Z+of CNC shield.
It should be the same as testing Z+, but without no disturbance of external elements.

You can also try to re-flash your arduino board with grbl firmware. I suggest use LaserGRBL to flash firmware (Menu "Tool")

StuartB4, 2020-12-01:
You are not trying to use the Z + for the PWM signal and the Z- for the limit switch are you. You can't use both pins for 2 different things. They are both connected together on the shield circuit. 

Also, do you have $32 set to 1 in grbl configuration on the grbl menu. If not you will get the burnt lines when the laser moves over blank / empty space in the image.

jmdang17, 2020-12-01:
> If you have 4.66v on D11 pin, also when you send M5, your arduino is defective.

Tested, it's showing 4.66v on D11 pin. :(

jmdang17, 2020-12-01:
> You can also try to re-flash your arduino board with grbl firmware. I suggest use LaserGRBL to flash firmware (Menu "Tool")
[image]
 
After I connected to Arduino board, I can't select the flash grbl firmware.

jmdang17, 2020-12-01:
> You are not trying to use the Z + for the PWM signal and the Z- for the limit switch are you. You can't use both pins for 2 different things. They are both connected together on the shield circuit. [...]

Hi StuartB4, what do you mean by use the Z+ for limit switch as well? 
Yes, I did set the $32 to 1 in grbl.
[image]

arkypita, 2020-12-01:
> > You can also try to re-flash your arduino board with grbl firmware. I suggest use LaserGRBL to flash firmware (Menu "Tool")
> After I connected to Arduino board, I can't select the flash grbl firmware.

You should not connect to arduino board.

jmdang17, 2020-12-01:
> > > You can also try to re-flash your arduino board with grbl firmware. I suggest use LaserGRBL to flash firmware (Menu "Tool")
> > > After I connected to Arduino board, I can't select the flash grbl firmware.
> 
> You should not connect to arduino board.

Hi arkypita, Ok, I will try again. So if it’s still not working, the only way the fix the current situation is to change a new arduino board right?

StuartB4, 2020-12-02:
` what do you mean by use the Z+ for limit switch as well?`

What I meant was, you cant use the Z+ for the PWM signal and use the Z- for a limit switch. But if you want to use the Z lmit switch you have to connect it to the SpnEn pin on the shield. 

I just mentioned it because at the very beginning of your post you wrote this. 

` I have another issue where once I connected any of the end stop limit on the board, the alarm will comes out and I can't do anything for it to run. If I disconnected all of the end stop limit switch from the cnc shield, everything is ok.`

jmdang17, 2020-12-02:
Hi StuartB4, now I got your point. My machine doesn’t has z axis so I didn’t use any z limit switch.

So if I change a new arduino board, the first thing that I will need to do is to check the voltage on the D11 pin as mentioned by the arkypita, without any external power source, am I correct?
If it’s showing 0v, means the arduino board can be use?

StuartB4, 2020-12-02:
Yes but you will need to put grbl on to the new one first before you test it. It will arrive without any firmware on it.

jmdang17, 2020-12-03:
Ok, I have another 2 arduino boards in hand, both are tested. When I checked the voltage on D11 pin, one board is 0v, another one is 4.66v. This is weird, so many boards are defective.
I will test the laser on the 0v arduino board and see how it goes.

StuartB4, 2020-12-03:
Are they genuine Arduino or clones? 
I use a genuine one but also have a cheap chinese clone. I used it a couple of times but keep it just in case I need it.
I also use the one with the plug in chip, then if it goes wrong I can swap the chip a couple of £'s.
I had to do it once. 

Ebay have them with the bootloader installed. 
https://www.ebay.co.uk/itm/2-5-10PCS-ATMEGA328P-PU-Microcontrolle-r-IC-Chip-With-ARDUINO-UNO-R3-Bootloader/142852729771?hash=item2142afabab:g:Sc8AAOSwUq9bO2XF

jmdang17, 2020-12-05:
Not sure whether they are clones or not.
I have tested the new board, it's working!!!  :)
But the end stop issues still persist, whenever I connect any of the limit switch to the board, the alarm will comes out and everything stalls, how can I solve this issue?

arkypita, 2020-12-05:
> But the end stop issues still persist, whenever I connect any of the limit switch to the board, the alarm will comes out and everything stalls, how can I solve this issue?

First of all, be sure you have read this document: https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches

By picture you post you have homing disabled and hard limit enabled (and it is ok for now). The behavior you have to expect is that when you switch on the machine is in Idle and that, only if you press one of the limit switches, the machine goes into Alarm.

If the machine goes to alarm when you turn on is it because the limit switch is sensed as "closed" (even one impulse can be enough).

This could be due to incorrect wiring, electrical noise, or incorrect configuration.

First of all you should to understand if your limit switch are NO or NC.
Default is NO, if you have NC switch you should invert limit pin configuration ($5=1)
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#5----limit-pins-invert-boolean

arkypita, 2020-12-05:
picture of your switch and wiring could help

jmdang17, 2020-12-06:
Thanks arkypita for the info, I belief I'm using the NO limit switch. Attached my end stop pictures and wirings.
I'm using 2 of these end stop for both end of my x-axis. 
I followed some info online, connected only the black and green wires and left the red one out, and instead of connect to the Z+ on the board (as per picture), I connected mine to X+ and X- on the board. 
Actually I have connected the end stop like this before and it's working and I'm not sure why this time it isn't.

[image]
[image]

jmdang17, 2020-12-06:
Here are my y-axis limit switch, I have 4 of these for both y-axis rails.
[image]
[image]
[image]

closed 2020-12-16 by arkypita