arkypita/LaserGRBL #1182: M4 option disappearing between runs

MarkDowes, 2020-11-30:
Hi I am having an issue where the M4 option is vanishing from existence between runs.

I launch LaserGRBL, I set up a project, i choose M4, i run it, no problems, then I go to run it again, and it has switched to M3 and M$ has disappeared altogether. I am chaing nothing, not even relaunching the app, just reloading last file and M4 is not available when it was moments ago.

Sometimes resetting everything and relaunching the app will bring M4 back, sometimes it does not reappear and I fiddle around till it magically reappears again.
This is happening both with loading vectors for cutting, and with leading images for engraving.

Have wasted some money and materials in the process from over-burning because it has switched to M3 and I have not realised, unfortunately.

arkypita, 2020-11-30:
http://lasergrbl.com/faq/

arkypita, 2020-11-30:
You say you don't change anything. Maybe some custom button you have is changing $32 laser mode option.

MarkDowes, 2020-11-30:
The only custom button I am using is the "turn on laser for focusing", none of the others. I see in editing that this changes S32 to 1 and 0, its also firing in M3 mode, would this effect it?

If so how is it that turning everything off, unplugging, restarting the computer, relaunching the app, does not make it reappear sometimes, but the other times it does?

arkypita, 2020-11-30:
> I see in editing that this changes S32 to 1 and 0, its also firing in M3 mode, would this effect it?

Yes. 

I suggest you change like this:

[image]

arkypita, 2020-11-30:
> If so how is it that turning everything off, unplugging, restarting the computer, relaunching the app, does not make it reappear sometimes, but the other times it does?

You can turn on off PC, app, unplug etc but $ parameters are stored permanently inside the control board so they don't change.
LaserGRBL read them at first connection and when you open the edit grbl configuration dialog.

MarkDowes, 2020-11-30:
Aaah ok, thank you very much for your help.

As always thank you for your continued development.

closed 2020-12-16 by arkypita