arkypita/LaserGRBL #1183: Sharp (Nearest Neighbour) causing checkboard error

MarkDowes, 2020-12-01:
Hi. A new day a new query.

When importing an image and selecting "Sharp" (Nearest Neighbour)" for the resizing parameter, the final result has bits missing from it in a checkboard fashion. Images attached. Anything I can do to prevent this?

[image]
[image]

The input image is a 1bit dithered bitmap.
I am trying to size my image in photoshop so that it is not rescaled at all by LaserGRBL.
It is very confusing converting between ppcm, ppi, 300 ppi, 72 ppi, and now downscaling to 30ppcm, my brain is in tangles.
(If am burning at 3 lines/mm = 30ppcm, 10x10cm image should be 300x300px, right?)

As a request, or two, an option to disable image resizing, and displaying some info about the size the image is, and the size it will be one engraved, would be very helpful for me, just to glance at and make sure everything is as expected before hitting that play button.

Thanks for your time!

arkypita, 2020-12-03:
I think that pre-dithered image cannot be processed by LaserGRBL because LaserGRBL scale the image with the artifact you see.

Yes I understand the request of a "bypass processing". I add to development roadmap #64 cannot say when I can add the feature.

arkypita, 2021-01-23:
https://github.com/arkypita/LaserGRBL/releases/tag/v4.1.1

arkypita, 2021-01-23:
Let me know

closed 2021-06-27 by arkypita