arkypita/LaserGRBL #1184: Scaling S-Min S-Max to image grays.

MarkDowes, 2020-12-01:
Hi again.

So I am wondering if LaserGRBL scales the S-Min/S-Max values to they grey values in an image?

[image]

I am struggling to get the light tones of the top image, and the dark tones of the bottom image, all in one engrave.
I can have it all heavy and dark, where the lighter areas are also very dark and the white areas are not burned at all, or, I can have it all very light as above, but the darker areas are very light and do not burn heavily.

No matter how much I play with contrast/brightness, its always one way or the other, and never a nice blend between the lights and darks.

I am playing with the S-min so that lighter areas are not completely ignored, but I am wondering if it is just "clipping" the S-min so that anything below the S-min value is ignored and just burned at S-min, or if the entire output is scaled between the S-min/S-max values.

If not, such a scaling option, I believe, would be a great feature.

I hope this makes sense.

Thanks for your time!

arkypita, 2020-12-01:
> So I am wondering if LaserGRBL scales the S-Min/S-Max values to they grey values in an image?

SMax is used for black
SMin is used for white

Any gray value between 0-255 is linear scaled between SMin-SMax

> I am struggling to get the light tones of the top image, and the dark tones of the bottom image, all in one engrave.

This is because material does not linear react to the laser power. Especially the plywood has a threshold below which it remains white, and above which it burns without too many possibilities of shades.

For these materials there is the dithering mode, where white / black points are alternated in such a way as to optically emulate the presence of a gray scale.

MarkDowes, 2020-12-01:
Hmm ok. Good to know. 
I understand that they ply has a minimum power to react that is why I ask. 
I am getting wildly inconsistent results which lead me to believe it might have been the scaling issue, but, alas, no.

Even with the dithering I am not getting the results I am wanting or expecting, and I cant seem to replicate results when I write down settings I used for an engrave, go back to those settings later and try again, different results.

Being able to save user pre-sets on the import screen might be helpful, and including the "target image" settings in the user pre-sets too. 
So many settings to fiddle around with, it is hard to keep on top of it all, especially switching between projects and images and having to change all the settings manually.

As always thank you for your time.

closed 2020-12-16 by arkypita