arkypita/LaserGRBL #1185: strumento di conversione foam cutter guardare foto

sasakingsoft, 2020-12-03:
[image]

does anyone know how to add that foam cutter component? where there are yellow dots

arkypita, 2020-12-03:
where does the image come from?

sasakingsoft, 2020-12-03:
From a youtube video

sasakingsoft, 2020-12-03:
Do you think it is possible to do it?

arkypita, 2020-12-03:
It is a customized version of lasergrbl, for use with a polystyrene cutter.
It is supplied with the machine.

sasakingsoft, 2020-12-03:
pensi non ci sia modo per usarlo o altre opzioni ? 
Do you think there is no way to use it or other options?

sasakingsoft, 2020-12-03:
e possibile avere una conversazione con te ? da qualche parte ? non uso molto bene github 
Is it possible to have a conversation with you? Somewhere ? I don't use github very well

arkypita, 2020-12-03:
mandami il tuo numero su info@lasergrbl.com