arkypita/LaserGRBL #1187: Cutting and engraving at the same time

MekoZeko, 2020-12-04:
Hello. I'm very new to laser engraving. Is that possible to engraving and cutting at the same time? Like engraving something and cut the outline border. I searched about this and didn't find anything.

Thanks.

nitro912gr, 2021-01-03:
I would love to know as well if it is possible with LaserGRBL :)

arkypita, 2021-01-03:
LaserGRBL is a simple gcode generate and stream, without advanced CAD/CAM features.
Nothing prevents these functions from being implemented in the future, but at the moment it is not possible.

Ps. you can always create your work in several parts and load them with different settings via the "file - append" menu

closed 2021-01-03 by arkypita

khturnings, 2021-01-03:
Not currently possible as far as I know but I do it by creating 2 files. One of the image and one of the outline. Then set up the raster engraving of the image. Then use the append file option to do the outline, making sure you select the number passes to use to do the cut.
Then save if you want to repeat.

khturnings, 2021-01-03:
oh - @arkypita got there first 👍

sergejzr, 2021-10-19:
Hello @arkypita , hello guys!
Is it possible to do the procedure (open file 1 with settings A, append File2 with settings B, save as gcode) from the command line? I need to convert a number of files this way. Also think this would be useful procedure for other users. Making it manually is a kind of nightmare :)

PS: To give some context, I am creating drawings for RS airplanes and ships for my father.
What I am doing is:
-  I have a large SVG file, where all parts are contained. For each engraving page there is a layer for cutting and a layer for text
-  I have a script that splits this large file into separate svg files, orders them in folders according to Layers names

Input - large.svg
Output: 
- myCoolAirplane
- - wood5mm
- - - text1.svg
- - - cut1.svg  
- - - text2.svg
- - - cut2.svg  
- - wood3mm
- - - text1.svg
- - - cut1.svg 

what I have at the end, for each engraving page, are two files (text/cut)
What I like to have is for each page one gcode that combines both having different laser strength.

sergejzr, 2021-10-31:
I just implemented a first version (python) script for converting multilayered SVG into GCode. It is possible to specify parameters (like speed, laser strength, repetitions) for different hierarchical layer names in SVG. Also, the script can automatically leave support gaps - parts on the path that are cut with less density and are used just to hold the piece.
An example input would be an SVG file with the following layer structure:

- myCoolAirplane
   - Layer name="Wood"
       - Layer name= "01"
         - Layer name="Engrave"
             - paths
             - ...
         - Layer name="Cut"
            -  - paths
             - ...
 

Example output would be a GCode like - "For material 'Wood', "Engrave" first with strength 400 and speed 200, then "Cut" 5 times with speed 20 and strength 900 leaving a support gap  of 1 mm" 

If you like, you can check it out here: https://github.com/sergejzr/svg2gcode_scripts. Would be a good idea to integrate the functionality directly into LaserGRBL

ragesoft, 2023-10-13:
Hi There. Thx for the video and the new feature @sunnysonic  .
I have tried it out and it worked, but i realized, that the engraving is also running 4 times, like my cuting.
Is there any way to set:
Import first file => set speed, power, run count (e.g. 1000, 5%-80%, 1 time)
Import / Append second file => set speed, power, run count (e.g. 300, 80% constant, 5 times)

Right now my workaround is to increase the speed of the engraving part, so it passes also 5 times, but will not do to much of engraving each time... However, this is very time-consuming.

:)

sunnysonic, 2023-10-14:
@ragesoft it works exactly the way you've asked. please watch the video again. it is described there. you add the first file, do all your settings, then append the second file and do your settings for that one.