arkypita/LaserGRBL #1188: laser doen(t worl

jean33-cpu, 2020-12-06:
hello , 
Is it normal for the machine to make this noise during the
end of the homing cycle? 
and too, the probrem is take a  very long time , the buffer stay running and can't start.
too , when is start,
the laser test command works but the lazer does not work anymore during the program phases.
is this problem due to the fact that my computer has 4 GB RAM?
Thank you for your help.

gmmanonymus111, 2020-12-07:
> Is it normal for the machine to make this noise during the
> end of the homing cycle?

Well, it is normal for your cooler fan to make noise.  Other than that, the homing cycle clicks your end switches 2 times (by default), the first one faster, the second one a bit slower to properly get it's position. No other noise should come from your machine as far as I know.

> and too, the probrem is take a very long time , the buffer stay running and can't start.

I don't quite get that

> too , when is start,
> the laser test command works but the lazer does not work anymore during the program phases.
> is this problem due to the fact that my computer has 4 GB RAM?

What is the GCode (program) you try to send to your machine? What is your `laser test command`?
Your amount of RAM should not be a problem. If LaserGRBL would try to use more than what is available, the OS would shut it down, with an error message.

jean33-cpu, 2020-12-07:
Hello , thank you for your reply . 

Can I send you the video for the Noise?

I just start with this machine  , and so , I don't understand : What is the GCode (program) you try to send to your machine? What is your laser test command?

Best regards. 

Jean

jean33-cpu, 2020-12-07:
I use the ortur master 2 with the software lasergrlb

gmmanonymus111, 2020-12-07:
Sure, you can send me the video to gmmidleacc1(at)gmail(dot)com

By GCode, I meant the engraving file (which is generated when you import an image or vector file), and can be seen here while sending it to your engraver
[image]
You can save it to a file (for later upload) by clicking `File`->`Quick save`
[image]
Since you wrote the laser isn't working during program phases, I assumed you sent GCode

gmmanonymus111, 2020-12-08:
On the video you've sent me it sounded like your stepper motor was running. The video was too short, so I'm not 100% sure. One thing I'm certain about, it seems a hardware issue, so I advise you contact Ortur support [here](https://ortur.tech/support/) or at their [Facebook page](https://www.facebook.com/groups/orturusergroup/)

jean33-cpu, 2020-12-08:
Hi, 
thank you again for your  replay; I fix  my noise problem ( wrong connection )but now  , there is no noise , and the home process is working properly , the program is runnig properly , but since, there is not lazer working, only the fan is working  :-(
I ll send the video with wetranfer , because to big size....
I dont now why?
BR

jean33-cpu, 2020-12-09:
Hello , ths is 2 video I made to show you my issue :
https://www.youtube.com/watch?v=m4jkpxNa5kU

https://www.youtube.com/watch?v=8r1VA3PoQvs

Have a nive evening.cdllt

    Le mardi 8 décembre 2020 à 21:48:28 UTC+1, Jean Prevot <dora6604@yahoo.com> a écrit :  
 
  Hi, 
thank you again for your  replay; I fix  my noise problem ( wrong connection )but now  , there is no noise , and the home process is working properly , the program is runnig properly , but since, there is not lazer working, only the fan is working  :-(
I ll send the video with wetranfer , because to big size....
I dont now why?
BR

gmmanonymus111, 2020-12-09:
Just for testing, if you type `G1F1M4S1` and then `G1F1M4S1000`, which one produces brighter laser light? It might be your laser driver needs inverted TTL signal. (As your GCode seems to send S1000 on the second video)

jean33-cpu, 2020-12-09:
Hello , i will check it , I dont know this evenin g i ll try to check you adwise . BR

closed 2020-12-16 by arkypita