arkypita/LaserGRBL #1189: Unexpected Disconnects at random times

gxgam, 2020-12-06:
I am getting UnexpectedDisconnect at seemingly random times. I've been trying to troubleshoot the problem the problem by using different settings like setting it to synchronous under streaming mode and slow under threading mode but still get the disconnects. The machine does use a cheap chinese arduino clone with the ch340g chip so I might try switching it out with an official version but I thought I'd get some advice first. 

In the log I get:
12/4/2020 3:38:17 PM.379	Issue detector	UnexpectedDisconnect [9,-1,-1]
12/4/2020 3:38:17 PM.380	SendImmediate	System.UnauthorizedAccessException: Access to the port 'COM5' is denied.
12/4/2020 3:38:17 PM.380	SendImmediate	   at System.IO.Ports.InternalResources.WinIOError(Int32 errorCode, String str)
12/4/2020 3:38:17 PM.380	SendImmediate	   at System.IO.Ports.SerialStream.EndWrite(IAsyncResult asyncResult)
12/4/2020 3:38:17 PM.380	SendImmediate	   at System.IO.Ports.SerialStream.Write(Byte[] array, Int32 offset, Int32 count, Int32 timeout)
12/4/2020 3:38:17 PM.380	SendImmediate	   at System.IO.Ports.SerialPort.Write(Byte[] buffer, Int32 offset, Int32 count)
12/4/2020 3:38:17 PM.380	SendImmediate	   at LaserGRBL.ComWrapper.UsbSerial.Write(Byte b) in E:\LaserGRBL\LaserGRBL\ComWrapper\UsbSerial.cs:line 124
12/4/2020 3:38:17 PM.380	SendImmediate	   at LaserGRBL.GrblCore.SendImmediate(Byte b, Boolean mute) in E:\LaserGRBL\LaserGRBL\GrblCore.cs:line 1141
12/4/2020 3:38:17 PM.397	SetStatus   	Machine status [Disconnected]

Tried it with a powered USB hub and would get just the UnexpectedDisconnect error. Also tried a couple USB cables, both with ferrite chokes.

Any help would be appreciated. Thanks!

gxgam, 2020-12-11:
Update:

I added an official arduino nano board. I still get UnexpectedDisconnect at random times but with nothing else logged after. I tried monitoring the serial port while the machine was etching and I'd get an UnexpectedDisconnect from lasergrbl but I wouldn't see anything out of the ordinary on the monitor. The monitor would still be connected but lasergrbl would disconnect.

Now I'm trying some other software and it's been etching fine for many hours so far. But I'd really like to figure out the problem with lasergrbl.

arkypita, 2020-12-11:
Hi @gxgam and thanks for your feedback

I have tested the system with many hours of work, also LaserGRBL is used by 50K + of people, which is the best test you can do.
In any case, this problem exists, and I would like to be able to solve it once and for all.

The problem with these exceptions is that they are outside the call stack of my program, that is, they occur within the libraries that microsoft makes available to me to communicate with the serial port (.NET Framework).

I am thinking of writing a serial access wrapper other than the microsoft one, to see if with direct access I can solve these problems, before this (it will take a few weeks or maybe more) you could do some tests.

- try to install / re-install the microsoft .net framework (google .net framework)
- temporary disable your antivirus when engraving (maybe is the antivirus that keep the port opened)

Let me know if you have some news

gxgam, 2020-12-11:
Thanks for the reply. I'll try reinstalling the .net framework plus that gives me some other ideas to test out. I'll update if I find anything new.

closed 2020-12-16 by arkypita