arkypita/LaserGRBL #1191: Laser Issue / Broken?

Iangwm, 2020-12-07:
Hi,

Sorry this isn't technically a issue with LaserGRBL, so delete if off topic.

I am a Laser Noob, just bought a cheap (might be my problem) diode laser kit off ebay , like this [one](https://github.com/arkypita/LaserGRBL/issues/1156). Just to get me going and a bit of experience.

I have got everything working and can cut paper, yeah. However when
[image]
 the laser 1st turns on it doesn't appear to be at full power. The 1st second of the cut it seems to be on low power then the power comes on and it starts cutting. It does this every time the laser is started.

I have reviewed the G Code and there is only one power setting at the start S800 and then the moves and laser on and off appear to be in the correct positions. So I am pretty sure it is a problem with the laser or the control board.

Just wondering is anybody else has had seen this issue and if they had fixed it - other than replace the laser.

I am in contact with the seller but they are not being that quick with their responses. 
I prob will end up returning it and getting an Ortur.

PS Laser GRBL is really good - I downloaded and got my cheap kit moving and laser almost working really quickly with the minimal of fuss. Cheers for your hard work.

Thanks
Ian

StuartB4, 2020-12-07:
Have you tried using M3 to cut. It's in Target image settings.

arkypita, 2020-12-08:
White material is difficult because reflect the laser color. Did you try dark color?

closed 2020-12-16 by arkypita