arkypita/LaserGRBL #1194: Cutting with the laser

SassyLady59, 2020-12-08:
Hi! I am totally frustrated!! I have tried so many different settings (suggestions from Ortur and CNC3-3018Pro) , trying to cut through 1,6mm plywood. For example: S1000 with 200mm/min and I should only need one or two passes. After 4 passes and at different sppeds I just burn the wood on the top and nothing goes through. What am I doing wrong?

I can engrave. The picture turned out fine, so I think I focused the laser correctly.

Please help me!!

Thank you!!

Annette

arkypita, 2020-12-08:
What kind of laser do you have? Power?

SassyLady59, 2020-12-08:
Thank you rfor the speedy reply...
I have a 15 w Ortur
For the wood cutting I used 1000 power and 200 mm/ min

SassyLady59, 2020-12-08:
I just tried to cut through again, to no avail. You can see little scratches on the back side, but not enough so that the pieces of wood pop out.
[image]
[image]
[image]
[image]

arkypita, 2020-12-08:
First, do not use vectorize for this kind of image. Use centerline instead.

Vectorize is ok with filled figure like this one
[image]

But will produce internal and external border (double line) if used on line-make images.

arkypita, 2020-12-08:
Second: i have ortur too (15 and 20W) there isn't much difference between the two and I honestly prefer the 15W.

With S1000 and F400 i cut poplar plywood up to 5mm in 8-10 passes BUT it is a very light plywood. Other essences and other densities require longer times, the glues used in the production of plywood also greatly influence.

Focus is the key to everything. It must be extremely perfect.
This look focused, other pictures does not look focused.

[image]

To achieve optimal focus you need to turn on the laser at the lowest possible power and focus on a non-reflective black material. the ideal is to glue a small piece of black electrician's tape on the plywood to be cut, and focus on it.

The 15W model has an optimal focus distance between 70-80mm

[image]

arkypita, 2020-12-08:
This means that it focus better if you have 7-8cm between your plywood and the laser module

SassyLady59, 2020-12-09:
Thank you so much. I will try it out on Friday & will let you know what happens!

closed 2020-12-16 by arkypita