arkypita/LaserGRBL #1195: laser stays on during cooling

Bazza1959, 2020-12-08:
I have an issue where my laser stays on during cooling, it all depends on when the cooling starts. If the laser is on when cooling starts it stays on, but if it was off, in say travel, then it stays off during cooling. What can I do to fix this as I had a print get spoiled yesterday because the laser stayed on during a supposed 5 minute cooling period which left the laser on in one spot for 5 minutes and burnt a horrible big hole in my project

arkypita, 2020-12-08:
Auto cooling only will work with certain condition. Are you sure you respect them?
There is a visible warning message in the box where the feature is activated.

[image]

Bazza1959, 2020-12-09:
I must apologize for opening this issue, as after I did, I then found the answer you have shown above

closed 2020-12-09 by Bazza1959