arkypita/LaserGRBL #1204: Starting point in the centre on round workpieces

OllieBFG, 2020-12-13:
Hi,
Does anyone know, if I can set the zero / home location to the centre of a workpiece, in order to have the engraved picture equally distributed on the rim of a plate for example? Zero is always in the bottom left corner and if I measure my workpiece only slightly incorrect, it is not central anymore. 

The attached picture is a sample (not mine) and it's also not centred. A slight mis-measurement or mis-calculation would shift the design off-centre and the workpiece is ruined. 
[image]

Any help would be greatly appreciated.

Thank you,
Oliver

BeatSlayer, 2020-12-13:
There's a centering helper already implemented, it's at the power/speed/size gcode creation dialog, it's a small bull's-eye.

OllieBFG, 2020-12-13:
Thanks for the fast response. I haven't noticed that yet. Will look again
and see, if I can find it.

Thanks very much for that. It's really appreciated.

Oliver

BeatSlayer, 2020-12-14:
From the releases docs:

[image]

closed 2020-12-16 by arkypita