arkypita/LaserGRBL #1206: Store "$ 32 = 1" permanently in memory possible?

rchobby-net, 2020-12-14:
Hello, I have the problem that after every restart of LaserGRBL is "$ 32 = 0". Is there a way to store "$ 32 = 1" permanently in memory?

StuartB4, 2020-12-14:
It's stored in the chip on the main controller board. It should'nt change on it's own. What board does your engraver have.

rchobby-net, 2020-12-14:
I have a Vigotec VG-7 Engraver. The chip is a ATmega 328.
[image]

arkypita, 2020-12-15:
You have another program that you use, that change the setting

or

you have some lasergrbl custom button that change this setting

rchobby-net, 2020-12-15:
StuartB4 & arkypita thanks for the replay.

I work only with LaserGRBL.
I checked the custom buttons. I have no button with any "$x= " commands.
I don't use gcode from other programs.

Maybe it would help to get a hexfile with preprogrammed "$ 32 = 1"?

arkypita, 2020-12-15:
It's a really bizarre behavior, the configuration values are permanently stored inside the non-volatile memory of the atmega328 chip. 

You have explicitly tested that if you enter lasergrbl, only connect and edit config, set $32=1 (look at the "success" message and re-read to have a confirm that the read value match) and exit lasergrbl, shut down the machine (and disconnect from USB) and then re-connect and read config, your $32 is = 0?

If so this is another Vigotech bullshit.

arkypita, 2020-12-15:
Also check your "startup blocks" https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#n---view-startup-blocks

rchobby-net, 2020-12-15:
I also installed lasergrbl on a stand-alone Arduino. It works there. I also suspect that there are still remnants of the vigotech firmware on the chip.

The "startup blocks" I will check at next.

rchobby-net, 2020-12-15:
$N gives:
$N1=
$N2=

Is it allowed to use these commands or can I get more problems:
$RST=$, $RST=#, and $RST=*- Restore Grbl settings and data to defaults

arkypita, 2020-12-15:
You can, but I don't think it change your behavior.

arkypita, 2020-12-15:
You are not the only one with this issue on this controller:

https://forum.lightburnsoftware.com/t/newbie-trouble-connecting-vigotec-l7/21545/13
https://forum.lightburnsoftware.com/t/force-laser-mode/20016/9

rchobby-net, 2020-12-15:
I have now reloaded GRBL from Github (1.1.h)and not the GRBL version supplied by Vigotec (1.1.f). Now it seems to be working. I will continue to monitor this and will contact you again if it doesn't work.

rchobby-net, 2020-12-15:
Sorry, I forget to say thank you for the support.

arkypita, 2020-12-15:
You are welcome!

closed 2020-12-16 by arkypita