arkypita/LaserGRBL #1207: CustomButton.gz file missing

Bunny120818, 2020-12-14:
Hi, I have downloaded the LaserGRBL software but it doesn't come with the CustomButton.gz file, it's nowhere to be seen on my PC. 

Step 1:
I Launch LaserGrbl

Step 2:
I Right-click the button area in LaserGrbl -> click “Import Custom Buttons”
I first open the software then I go below to right-click, then I select import

CustomButton.gz file cannot be found

 I tried to remove the program and redownload a few times but it still doesn't help. what should I do? Thank you

arkypita, 2020-12-14:
https://lasergrbl.com/usage/custom-buttons/

Bunny120818, 2020-12-15:
Hi, Thank you for your reply. I did click on the download button but nothing happened, "there's no download" 
[image]

StuartB4, 2020-12-15:
It should be in the default download folder on your computer, unless you download to a different folder. 
It is there somewhere if you clicked the download link.

StuartB4, 2020-12-15:
I just got it and uploaded it to my pcloud account. You dont need to register or open an account, just download it. 

https://u.pcloud.link/publink/show?code=XZPA64XZb0YdRsvhbE8PzBIYLT63JpqgVTtk

StuartB4, 2020-12-15:
[image]

StuartB4, 2020-12-15:
I use Firefox browser. If I hovver the mouse over the Download button it shows the file at the bottom of the screen. When I click it it downloads, you don't get any options to save or cancel, it just downloads. 

[image]

Bunny120818, 2020-12-15:
Hi, I've tried opening all files on my computer and the download is nowhere to be seen, but the link you just sent works! OMG Thank You. Next is to figure out how to use it haha thank you for the follow-up, you're awesome 

[image]

closed 2020-12-15 by Bunny120818

StuartB4, 2020-12-15:
Good to know you got it working.  
Thanks for letting us know. :+1:

lcv-vr, 2023-11-18:
Hello
I just downloaded the custom buttons from the lasergrbl website and the same thing happens to me that was mentioned in this thread. It doesn't let me open the file either with WinRaR or WinZip... it tells me that the file is damaged. Could someone please correct the error or upload the file in another format? I use Windows 10

arkypita, 2023-11-18:
You don't need to unzip it, just load in LaserGRBL as it is described in download page.