arkypita/LaserGRBL #1210: Newest software update has put my laser out of commission

ShelliTr, 2020-12-16:
So up until the recent update my laser has worked fine on lasergrbl. Now that I have updated it will not work at all.

The motor is running, it is very random when it will actually connect.
When it will connect and I send the file it endlessly file transfers and never actually starts the burn.

I am at a complete loss for what to do, and I just had an order last night come in for some ornaments I make with my laser and I am starting to panic. Please help.

hixnhx, 2020-12-16:
I have great success with lasergrbl untill the last update and every thing i try to do just burns and doesn't cut.  how do I revert to previous  version

arkypita, 2020-12-16:
Are you talking about v3.8.5?

arkypita, 2020-12-16:
All LaserGRBL version are available here: https://github.com/arkypita/LaserGRBL/releases

closed 2020-12-16 by arkypita