arkypita/LaserGRBL #1211: laser engraver not homing and going past limits , this is the error message said to send to this site

Merlinshabbit, 2020-12-17:
[image]

arkypita, 2020-12-18:
Thanks for the report, is it something that happens always or sporadically?

The problem has already been reported to me, but it seems to be a problem with the Microsoft serial management libraries over which I don't have much control.

Merlinshabbit, 2020-12-18:
This was first time . When I was taking some pix of plugs and moved the
laser head to the middle to dona video of it turning on , it homes
correctly all of sudden. I was asked by a support email I reached out to,
tondo this and send it to them . Came to conclusion that it may of been a
plug coming loose from movement and my zip tied were not right enough. But
next project my laser works then stops burning 26% in to project, but kept
moving . I tried again and worked . Curious about this Microsoft serial
issue. Do you think I should have the buffer set to a lower setting? I
mainly use the engraver for wood burning images .