arkypita/LaserGRBL #1212: Getting Negative Values and After Homing

MekoZeko, 2020-12-18:
Hi. I'm getting negative values after homing. Homing is successfull but after I tried to engrave something it goes out of bounds. Image starts at (0,0) but homing is (-390,-10) and it try to go positive values.What should I do? Settings: [https://imgur.com/bPOv4mu](url) [https://imgur.com/2BXeklX](url) Homing left up corner.

arkypita, 2020-12-18:
I see your picture and settings.
Settings look ok.

So, you hit the homing button, the machine "sense" the limit switch, and after that you don't have a 0,0 position but you have -390,-10 instead?

This is an hardware issue, not a LaserGRBL issue/bug (and this is an help request, so please next time post under "discussions" section.

arkypita, 2020-12-18:
Is it behavior that started after you did something or has your machine been doing this since you have it?

You can always press the "globe with pin" button, to assign the current zero, but it would be better to understand and solve the problem at the root, because after homing the position must be 0,0 by definition.

PS. if you google "grbl wrong position after homing" you can found some different threads about this issue.

MekoZeko, 2020-12-18:
Thanks for your reply. My machine is Atomstack A5 30W. It didn't come with limit switches so I installed them afterwards. I think I installed them correctly because $23=1 is a top left homing and it's doing it with no problem. After homing it shows negative coordinates as home coordinates. As you said it's probably firmware problem. I search for this in google and it's common thing  thar working on negative space in CNC but since LaserGRBL works on positive space so I think I have to change firmware.

I think I will pass homing, because I'm new to laser engravers and don't know how to recompile GRBL. I have this engraver for 4 days and don't want to damage the machine :). Thanks for your help.

arkypita, 2020-12-18:
I also have an atomstack, but without limit switches, so i cannot test your issue.

I have found this link [1] that say:

> There’s a good chance that you have a workspace offset in place, which is common on CNC machines.
> Type $# into the console and hit enter. You’ll see a bunch of output like this:
> [G54:0.000,0.000,0.000]
> [G55:0.000,0.000,0.000]
> [G56:0.000,0.000,0.000]
> [G57:0.000,0.000,0.000]
> 
> Those are the workspace offsets. If they aren’t zeros that could be it. To zero them, type:
> 
> G10 L2 P2 X0 Y0 (and press enter)

[1] https://forum.lightburnsoftware.com/t/home-position-incorrect/6736

Did you try to type $# into "type gcode here" in lasergrbl interface and hit enter?

MekoZeko, 2020-12-18:
https://imgur.com/L8Q41q1 it's showing correct numbers. But still negative homing coordinates (right corner)

arkypita, 2020-12-18:
It's really strange. You can try to contact Atomstack, luckily they are a reliable company and I am convinced that they will answer you.

http://atomstack3d.com/lxwm

Otherwise you can try rewriting/flashing the firmware with a standard copy of GRBL but there is a little risk that atomstack need a customized firmware so this could break the machine http://lasergrbl.com/usage/flash/

MekoZeko, 2020-12-18:
Thanks for your help. I will contact with Atomstack. Homing is a not big deal for me but if works then that can be a plus.

arkypita, 2020-12-18:
> because after homing the position must be 0,0 by definition.

My mistake!

https://github.com/gnea/grbl/blob/eefe2bb95bb7b21ec2bb87e6ab6e20747e1626c4/grbl/config.h#L129

Standard GRBL settings.h (compiled in firmware) say that after homing the machine is set to negative space, so this is the standard behavior of a firmware.

To have 0,0 you need a firmware with `HOMING_FORCE_SET_ORIGIN` uncommented.

If you read my documents [here  ](http://lasergrbl.com/usage/flash/) lasergrbl give you a ready-to-flash firmware compiled for this option.

MekoZeko, 2020-12-18:
Can I flash firmware with that file directly in LaserGBRL or is there anything I need to change to work with Atomstack A5? Thanks.

arkypita, 2020-12-18:
Looking at the control board under the metal cover it looks like a standard board, but I can't guarantee it.

MekoZeko, 2020-12-18:
Should I choose Arduino Uno for target in LaserGRBL?

arkypita, 2020-12-18:
Uno or nano it't the same atmega chip, only change the programming baudrate speed.
Test Uno if not work try nano.

MekoZeko, 2020-12-18:
I flash firmware and it's homing to 0,0 now but x goes 0 to positive and y goes 0 to negative now.

arkypita, 2020-12-18:
You can invert axis direction via grbl configuration

MekoZeko, 2020-12-18:
I tried to do that and suddenly it disconnected from computer and dont show up again. What should I do?

MekoZeko, 2020-12-18:
Did i damage the board?

MekoZeko, 2020-12-18:
I opened the case and wait a bit then it works now. I didn't want to mess with that again :)

arkypita, 2020-12-18:
The connect issue is not related to firmware flash and opening the case is not related to its solution. Just casualty.

arkypita, 2020-12-18:
All is fine now?

MekoZeko, 2020-12-18:
Yes. I tried to engrave something and it's working fine. I guess I will wait for someone try to add limit switches for Atomstack A5. Thanks for your helps.

closed 2020-12-18 by MekoZeko