arkypita/LaserGRBL #1214: Timeout connection

PalladinoDaniele, 2020-12-18:
Hi!
I'm trying to connect laserGRBL with My sweety Laser, mounting a CH340 choosey. The problem is that connection goes to time out!

I'm using v3.8 and driver is 2019 year.

Have someone solved this issue?

Thank you in advance

Dan

arkypita, 2020-12-18:
This is an help request, not a software issue: it is more correct that it is placed in the "discussions" section.
Look like that this machine came with proprietary software, nothing in the amazon page indicates that it can be used with LaserGRBL. 

[image]

In order to be used with LaserGRBL a marker must be equipped with a control board with GRBL firmware. I have no idea what that machine uses as a control board and firmware.

Do you have any information about it?

PalladinoDaniele, 2020-12-18:
> This is an help request, not a software issue: it is more correct that it is placed in the (new) "discussions" section.
> 
> Look like that this machine came with proprietary software, nothing in the amazon page indicates that it can be used with LaserGRBL.
> 
> [image]
> 
> In order to be used with LaserGRBL a marker must be equipped with a control board with GRBL firmware. I have no idea what that machine uses as a control board and firmware.
> 
> Do you have any information about it?

In the amazon page is written IT CAN SUPPORT GRBL, check image below is italian, the sentence you have to translate is: "GRBL può supportare"

[image]

Any ideas?

PalladinoDaniele, 2020-12-18:
Anyway, this is the product to check:
https://www.amazon.it/dp/B0842KWP12/ref=cm_sw_r_cp_apa_fabc_UZk3FbETY5218?_encoding=UTF8&psc=1

But I want to mark that the software is working properly with its own software, i bought it because of GRBL to have more settings....

arkypita, 2020-12-18:
I google for mysweet laser and this one is the first result, but now I see that there is a lot of products with this brand (this is only a commercial name of re-branded anonymous product, not a real factory name). 

[image]

Questo che mi hai linkato sembra più una macchina grbl, anche se la foto del loro software cinese mi fa pensare male.

Comunque, se fai una registrazione della sezione di connessione (menu ? -> activate extended log e quindi dopo aver dato il nome al file premi su connect)  e poi me la carichi qui, posso dirti qualcosa di più.

PalladinoDaniele, 2020-12-18:
Ok, this moment I'm writing from cellphone, but the extended log has a long list of "3F"....I'll upload the log asap

PalladinoDaniele, 2020-12-18:
Extended LOG:

00000	00581471	log	Recording session started @ 17/12/2020 16.21.44
00001	00583764	com	Open COM3 @ 9600 baud Ctrl-X
00002	00584331	tx	[18]
00003	00584367	tx	[3F]
00004	00584889	tx	[3F]
00005	00585392	tx	[3F]
00006	00585888	tx	[3F]
00007	00586391	tx	[3F]
00008	00586888	tx	[3F]
00009	00587393	tx	[3F]
00010	00587891	tx	[3F]
00011	00588390	tx	[3F]
00012	00588888	tx	[3F]
00013	00589391	tx	[3F]
00014	00589892	tx	[3F]
00015	00590393	tx	[3F]
00016	00590891	tx	[3F]
00017	00591391	tx	[3F]
00018	00591890	tx	[3F]
00019	00592389	tx	[3F]
00020	00592888	tx	[3F]
00021	00593392	tx	[3F]
00022	00593767	com	Close COM3 [CORE]

arkypita, 2020-12-18:
9600 doesn't look like a standard speed for grbl engraver. Did you already try 115200?

PalladinoDaniele, 2020-12-18:
i've changed to 115200 to COM port and also in the software, but nothing happens, I've tried now:

00000	00030200	log	Recording session started @ 18/12/2020 14.37.27
00001	00031909	com	Open COM4 @ 115200 baud Ctrl-X
00002	00032525	tx	[18]
00003	00032560	tx	[3F]
00004	00033104	tx	[3F]
00005	00033606	tx	[3F]
00006	00034105	tx	[3F]
00007	00034606	tx	[3F]
00008	00035105	tx	[3F]
00009	00035605	tx	[3F]
00010	00036106	tx	[3F]
00011	00036606	tx	[3F]
00012	00037103	tx	[3F]
00013	00037606	tx	[3F]
00014	00038105	tx	[3F]
00015	00038604	tx	[3F]
00016	00039102	tx	[3F]
00017	00039603	tx	[3F]
00018	00040102	tx	[3F]
00019	00040606	tx	[3F]
00020	00041104	tx	[3F]
00021	00041604	tx	[3F]
00022	00041912	com	Close COM4 [CORE]

arkypita, 2020-12-18:
3F is a command sent by LaserGRBL to prompt the machine replying with a status report, but the machine does not send nothing and does not reply nothing (no rx lines).

A standard grbl session look like this:

```
00000	00060836	log	Recording session started @ 18/12/2020 14:44:54
00001	00062415	com	Open COM12 @ 115200 baud DTR, RTS, Ctrl-X
00002	00062418	tx	[18]
00003	00062421	tx	[3F]
00004	00062622	tx	[3F]
00005	00062823	tx	[3F]
00006	00062961	rx	\r
00007	00062962	rx	Grbl 1.1h ['$' for help]\r
00008	00062970	rx	[MSG:'$H'|'$X' to unlock]\r
00009	00063021	tx	[3F]
00010	00063030	rx	<Alarm|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:S|WCO:0.000,0.000,0.000>\r
00011	00063222	tx	[3F]
00012	00063230	rx	<Alarm|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:S|Ov:100,100,100>\r
00013	00063423	tx	[3F]
00014	00063430	rx	<Alarm|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:S>\r
00015	00063622	tx	[3F]
00016	00063628	rx	<Alarm|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:S>\r
00017	00063822	tx	[3F]
00018	00063829	rx	<Alarm|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:S>\r
00019	00064023	tx	[3F]
[...]
```

arkypita, 2020-12-18:
Can you try enable Hard and Soft reset in LaserGRBL settings?

[image]

PalladinoDaniele, 2020-12-18:
ok, so is not possible to understand the issue.....thanks a lot for your time!!!!!!!!!!!!!!

arkypita, 2020-12-18:
I downloaded their .rar file which is linked on the amazon page. There is nothing in the machine manual talking about GRBL, they only give examples of their software. I fear that the "grbl" in the amazon sale is a hoax.

arkypita, 2020-12-18:
Ho anche de-compilato il loro programma "Laser Framework2.exe" e da una occhiata veloce non sembra proprio che cerchino di parlare con una macchina GRBL, quindi confermo la mia ipotesi.

Teoricamente potresti tentare un flash del firmware grbl che andrebbe a sostituire il loro firmware proprietario MA se la scheda di controllo non segue i cablaggi e i pin classici di grbl ti ritroveresti con la macchina inutilizzabile e senza possibilità di tornare indietro (non mi pare che loro forniscano il firmware)

----

I also de-compiled their "Laser Framework2.exe" program and from a quick glance it doesn't look like they are trying to talk to a GRBL machine, so I confirm my guess.

Theoretically you could try a flash of the grbl firmware which would replace their proprietary firmware BUT if the control board does not follow the classic grbl wiring and pins you would end up with the machine unusable and with no possibility of going back (I don't think they provide the firmware for a rollback)

closed 2020-12-18 by arkypita