arkypita/LaserGRBL #1222: After install running LaserGRBL causes error and crashes

jaknoke, 2020-12-24:
LaserGrbl v3.8.6.0
 v
Wrapper: UsbSerial
Win 7, Ultimate, 6.1.7601.65536, Service Pack 1 (6)
CLR: C:\windows\mono\mono-2.0\lib\mono\4.5\mscorlib.dll 4.6.57.0

TypeOf exception  [System.TypeInitializationException]
Exception message [The type initializer for 'LaserGRBL.UsageStats' threw an exception.]
Exception source  [LaserGRBL], thread []
Exception method  [Void Main(System.String[])]

   ----------- stack trace -----------
  at LaserGRBL.Program.Main (System.String[] args) [0x00015] in <4b3130430cd14d049ed6acee3cb3d9d1>:0 

Inner exception data
TypeOf exception  [System.TypeInitializationException]
Exception message [The type initializer for 'LaserGRBL.GrblCore' threw an exception.]
Exception source  [LaserGRBL], thread []
Exception method  [System.String get_filename()]

   ----------- stack trace -----------
  at LaserGRBL.Logger.get_filename () [0x00000] in <4b3130430cd14d049ed6acee3cb3d9d1>:0 
  at LaserGRBL.Logger.DoTheWork () [0x00030] in <4b3130430cd14d049ed6acee3cb3d9d1>:0 

Inner exception data
TypeOf exception  [System.TypeLoadException]
Exception message [Could not load type of field 'LaserGRBL.PSHelper.MaterialDB:FromServer' (0) due to: Could not load file or assembly 'System.Data.DataSetExtensions, Version=3.5.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089' or one of its dependencies. assembly:System.Data.DataSetExtensions, Version=3.5.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 type:<unknown type> member:<none>]
Exception source  [], thread []
Exception method  []

arkypita, 2020-12-24:
I suggest you to install or re-install Microsoft .NET framework

arkypita, 2020-12-25:
Moved to [discussion](https://github.com/arkypita/LaserGRBL/discussions)

closed 2020-12-25 by arkypita