arkypita/LaserGRBL #1223: projected time question

ronisegoly, 2020-12-25:
Hi
I just replaced my device so it's brand new and I did not change anything in setting or board configuration
When I engrave I need to set the speed and I set it to 5000 mm/min
But when machine start moving the projected time is changed considerably ( like from 1 hour to 10)
Any idea?

arkypita, 2020-12-25:
Before start the time is estimated by set speed and total job lenght.
When running the estimated time is projected by the actual, real, speed of the engraver.

So, maybe your engraver is not moving at the speed you set in job.
This could be because you have a acceleration or max speed limit in your engraver configuration, so the engraver cannot reach the settled speed.

Please read: http://lasergrbl.com/wp-content/uploads/2020/05/Grbl-Configuration-ENG.pdf

arkypita, 2020-12-25:
However this is not a LaserGRBL issue, so it should be posted under "[discussions](https://github.com/arkypita/LaserGRBL/discussions)" (moved)

closed 2020-12-25 by arkypita