arkypita/LaserGRBL #1233: ERRORE IMPREVISTO DI COMUNICAZIONE

onofrio73, 2020-12-27:
LaserGrbl v3.8.7.0
Grbl v0.9j
Wrapper: UsbSerial
Win 7, Home Premium, 6.1.7601.65536, Service Pack 1 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v2.0.50727\mscorlib.dll 2.0.50727.8806 (QFE.050727-8800)

TypeOf exception  [System.ObjectDisposedException]
Exception message [L'handle Safe è stato chiuso]
Exception source  [mscorlib], thread []
Exception method  [IntPtr SafeHandleC2NHelper(System.Object, IntPtr)]

   ----------- stack trace -----------
   in System.StubHelpers.StubHelpers.SafeHandleC2NHelper(Object pThis, IntPtr pCleanupWorkList)
   in Microsoft.Win32.UnsafeNativeMethods.GetOverlappedResult(SafeFileHandle hFile, NativeOverlapped* lpOverlapped, Int32& lpNumberOfBytesTransferred, Boolean bWait)
   in System.IO.Ports.SerialStream.EventLoopRunner.WaitForCommEvent()
   in System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   in System.Threading.ThreadHelper.ThreadStart()

onofrio73, 2020-12-27:
[image]

arkypita, 2020-12-27:
Both are not LaserGRBL issue.

The first is a Microsoft .NET framework issue, check you have installed the [last version](https://www.google.com/search?client=opera&q=.net+framework&sourceid=opera&ie=UTF-8&oe=UTF-8)

The second (alarm status) is telling you that the machine is in alarm state because you have not done Homing or Unlock, and it is a [grbl state](https://github.com/grbl/grbl/issues/920).

closed 2020-12-27 by arkypita