arkypita/LaserGRBL #1238: cant connect

patriot1467, 2020-12-31:
i cannot connect my Lazer burner to my computer , i have tried all com ports , uninstalled program and reinstalled still to no avail , i even pressed the reset button on the controller , i even tried connecting to my friends laptop, i purchased this in March 2020 so its not that old , was working fine up until a month ago then started having problems with the burnings , can someone please advise me what could be wrong ,

arkypita, 2020-12-31:
Well, it look like an hardware issue. So why don't you write to the seller or manufacturer of your engraver?

arkypita, 2020-12-31:
In the meanwhile, can you check if a serial/USB port appare in windows "device manager" when you connect your engraver?

[image]

arkypita, 2020-12-31:
In the meanwhile, can you check if a serial port appare in windows "device manager" when you connect your engraver?

[image]

patriot1467, 2020-12-31:
Thanks for your reply, I will contact supplier

closed 2021-01-12 by arkypita

patriot1467, 2021-01-12:
I wrote to suppliers ( Yora ) unfortunately they are not forth coming with
a solution and say it's outside warranty, only had equipment less than 9
months when issues started to occur , I will unfortunately look for another
solution or provider of better quality product and customer service

arkypita, 2021-01-12:
I am sorry for your experience with Yora, thanks for share this with us.
It's a shame they don't answer for a 9 month product, it doesn't advertise well.

I think you can save your engraver with a simple control board replacement. If you need help finding one, please share some pictures of the one it mount, so I can find if there are cheap replacement.

patriot1467, 2021-01-12:
Thank you, I will do that. , I may even look at upgrading the Lazer to a
better one also