arkypita/LaserGRBL #1239: Laser Master 2

Rangers135, 2021-01-01:
Hello, 

I got an Ortur Laser Master 2 recently and it hooked up to my computer and everything just fine, but tonight I tried to run the machine with a picture I emailed myself and inserted the file into LaserGRBL and framed the program. When I started to run the program, the laser became very dim and not burning the image. I thought it was a connection issue from one of the wires being lose but that wasn't it. I tried to run the program again after adjusting a couple of connections and still nothing. Eventually over the next fifteen minutes or so, the laser would just cut out completely but with all of the mechanics and motors still working. I am very new to using a machine like this and I'm hoping its an easy fix. Any and all help is greatly appreciated. Thank you in advance.

StuartB4, 2021-01-01:
I would ask on the official Ortur Facebook group. Loads of help there. 
https://www.facebook.com/groups/orturusergroup

gmmanonymus111, 2021-01-01:
It sounds like your laser died. It is easy to fix, you just have to replace.your laser module, but unfortunately it could be pricy. If you have warranty on your machine, contact Ortur support or as Stuart suggested, ask around their Facebook group

closed 2021-01-12 by arkypita