arkypita/LaserGRBL #1245: Ooops! Something went wrong. (Computer without COM port)

tomturican, 2021-01-04:
Thank you again for your great work. Today I installed your program on my second computer. There is currently no machine connected and therefore there is no active COM port.
Since version 3.8.10 I can no longer call up the settings. It is possible that the existence of a COM port is not checked correctly and an error is caught.
[image]

arkypita, 2021-01-04:
Thanks for bug report. I'll check the code and release a fix

arkypita, 2021-01-05:
Was not for the com port, it was because you never connect it to a grbl board.
I will soon release the fix.

closed 2021-01-05 by arkypita

arkypita, 2021-01-05:
Fixed in v4.0.2 https://github.com/arkypita/LaserGRBL/releases/tag/v4.0.2

commit 32762a8b (2021-01-05): Fix #1245