arkypita/LaserGRBL #1248: Preview drawing issue

arkypita, 2021-01-05:
**Describe the bug**
LaserGRBL preview draw a segment in wrong position, while the machine execute it correctly

**To Reproduce**
Steps to reproduce the behavior:
1. Load the attached .nc file

**Screenshots**
[image]

**Hardware and software configuration (please complete the following information):**
 - LaserGRBL version 4.0.1 (but i think all version)

**Additional context**
The same image, vectorized with different sizes gives no error.

[image]

[image]

Attached files:
- original image file
- generated gcode

[ras.zip](https://github.com/arkypita/LaserGRBL/files/5769275/ras.zip)

svenhb, 2021-01-06:
What a strange gcode file: almost all straight lines are G2 / G3 with very large radius. 
The bad line is `G2 X31.687 Y7.553 I-3281922.101 J-2205536.466`, where the I part is probably out of a range.

svenhb, 2021-01-06:
I had the same problem in GRBL-Plotter and found the problem in GraphicsPath.AddArc Method, which only accepts float.
The difference between double and float for this specific code line, for the AddArc parameters are:
```
  double: x1:-7236054.91985262  y1:-6159689.27685262  r2:7908325.68370525  
  float:  x1:-7236055           y1:-6159690           r2:7908326           
```

BTW: This gcode also causes error 33 in lines 1575, 1741, 1755 and 1771 in regular grbl.

arkypita, 2021-01-07:
Hi @svenhb 
Thank you for your quick feedback. It is a pleasure to see that I am not the only one interested in the LaserGRBL code.
At the moment I cannot deal with this bug, but your information will surely be useful to me, when I dig into.

> What a strange gcode file: almost all straight lines are G2 / G3 with very large radius.

It is the potrace vectorization output.