arkypita/LaserGRBL #1253: CH340 driver?

itow82, 2021-01-07:
USB-SERIAL CH340 
Windows cannot verify the digital signature for the drivers required for this device (Code 52)
i have atmostack A5 and win 7

arkypita, 2021-01-07:
[image]

Where does your driver comes from?

itow82, 2021-01-07:
hi
Thanks for the quick reply
driver come with the software

[image]

arkypita, 2021-01-07:
> driver come with the software

I think you mean the one you can install from LaserGRBL's "Tools" menu. Correct?

It is exactly the same I have on my PC and I have no issue.

itow82, 2021-01-07:
yes  from LaserGRBL's "Tools" menu

arkypita, 2021-01-07:
Seem to be a your pc issue.

Did you try google for it?

itow82, 2021-01-07:
[image]

StuartB4, 2021-01-07:
I use this one in Windows 7 if I need to install one. 
https://learn.sparkfun.com/tutorials/how-to-install-ch340-drivers/all

Right click it and click Run as Administrator. 

[image]

[image]

itow82, 2021-01-08:
@StuartB4  thank you very mach  
@arkypita thank you very mach

closed 2021-01-08 by itow82

arkypita, 2021-01-08:
Does stuart driver work for you?

itow82, 2021-01-08:
yes

arkypita, 2021-01-08:
@StuartB4 if you think this version is better i'll update the package inside LaserGRBL

@itow82 can you put a screenshoot of the driver property dialog, with the new driver? just to understand the difference in version, provider etc

itow82, 2021-01-08:
[image]
[image]

StuartB4, 2021-01-08:
@arkypita 

The driver it'self It seems to work fine and without any problems.

The only problem I've had with it is when closing it after installing the driver. 
It pops this error box up. Maybe because it's an old version, may be my computer.  
It does'nt affect how it works, I just don't know why it does it.
It does'nt do it everytime either. 

I just click "This program installed correctly" and it's fine.

[image]

itow82, 2021-01-08:
I had the same

StuartB4, 2021-01-08:
@itow82.

Is it still working ok?
I thought it might just be me.

arkypita, 2021-01-08:
As my version is newer, I will continue to keep this one.

StuartB4, 2021-01-08:
Ok @arkypita.
:+1:

itow82, 2021-01-17:
@arkypita @StuartB4 
Sorry for the delay in reply
I built a box with a smoke pumping system and an air nozzle
And now I have tried the machine and it works
I still need to work on the settings of power and speed for better results
Thank you very much

StuartB4, 2021-01-18:
No problem. Glad you got it working. 
Thanks for letting us know. 
:+1: