arkypita/LaserGRBL #1254: LaserGRBL crash/unfinished engraving

lasermyfaceoff, 2021-01-07:
So I have two problems occurring... I'm trying to engrave a greyscale image via LaserGRBL's dithering options. Approximately every 15,000 lines of G-code, the laser stops moving and stops firing. To begin the engraving where I left off I have to reset the laser and disconnect it from LaserGRBL and reconnect it. Every time I disconnect the laser in LaserGRBL via button in top left corner, I get the error message copied at the bottom of this post. This occurs if I abort the program before disconnecting or not. The error message closes LaserGRBL if I close out of it or click the abort option in it. After its closed LaserGRBL I can't restart the engraving where I left off. If you have any idea how to fix either problem, it would be greatly appreciated.  

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware and software configuration (please complete the following information):**
 - Windows 7 Home Premium
 - Ortur Master 2 (15W)
 - Grbl version 1.1f
 - LaserGRBL v3.8.0

**Additional context**
LaserGrbl v3.8.0.0
Grbl v1.1f
Wrapper: UsbSerial
Win 7, Home Premium, 6.1.7601.65536, Service Pack 1 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v2.0.50727\mscorlib.dll 2.0.50727.8806 (QFE.050727-8800)

TypeOf exception  [System.ObjectDisposedException]
Exception message [Safe handle has been closed]
Exception source  [mscorlib], thread []
Exception method  [IntPtr SafeHandleC2NHelper(System.Object, IntPtr)]

   ----------- stack trace -----------
   at System.StubHelpers.StubHelpers.SafeHandleC2NHelper(Object pThis, IntPtr pCleanupWorkList)
   at Microsoft.Win32.UnsafeNativeMethods.GetOverlappedResult(SafeFileHandle hFile, NativeOverlapped* lpOverlapped, Int32& lpNumberOfBytesTransferred, Boolean bWait)
   at System.IO.Ports.SerialStream.EventLoopRunner.WaitForCommEvent()
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart()

arkypita, 2021-01-07:
Well... this error has to do with problems with the microsoft serial protocol with which I have fought a lot but without success.

To try to solve it I rewrote the serial protocol, then I put a setting (only in the [latest version of LaserGRBL, currently pre-release](https://github.com/arkypita/LaserGRBL/releases/download/v4.0.2/install.exe)) that uses this new protocol. 

[image]

It could also be a hardware electrical noise or power problem. There is a whole tips section [on the FAQ page](http://lasergrbl.com/faq/) that EVERYONE should read. 

I suggest you proceed for single variations, changing only one factor at a time, so as to gradually identify the problem.

1) change to UsbSerial2 and do a lot of testing just by only changing this one. if solve, it was microsoft issue i tell you.
2) if not solve, try slow down LaserGRBL thread mode 
3) if not solve, try change USB cable
4) if not solve, try change power adapter
5) ....

lasermyfaceoff, 2021-01-07:
Thanks for the quick response, a poster on reddit is having a similar issue with the Ortur and he's using Lightburn.

https://www.reddit.com/r/OrturLaserUsers/comments/krjcko/error8_cant_complete_any_job_please_help/

I'll try the new protocol and let you know.

lasermyfaceoff, 2021-01-07:
I just ran a test with v4.0.2 in USBSerial2, unfortunately it did not fix my issue. The gimble stopped but the laser continued to burn full power. I attached the error received bellow after aborting the program and disconnecting the laser. The continue button was nice because it allowed me to close the error without closing LaserGRBL but when I attempted to reconnect to the laser, LaserGRBL detected an error and forced me to close it. I'll start testing the hardware and reply after.

LaserGrbl v4.0.2.0
Grbl v1.1f
Wrapper: UsbSerial2
Win 7, Home Premium, 6.1.7601.65536, Service Pack 1 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.7.3468.0 built by: NET472REL1LAST_C

TypeOf exception  [System.BadImageFormatException]
Exception message [Invalid access to memory location. (Exception from HRESULT: 0x800703E6)]
Exception source  [mscorlib], thread [Main Thread]
Exception method  [Void ThrowExceptionForHRInternal(Int32, IntPtr)]

   ----------- stack trace -----------
   at System.Runtime.InteropServices.Marshal.ThrowExceptionForHRInternal(Int32 errorCode, IntPtr errorInfo)
   at System.Runtime.InteropServices.Marshal.FreeHGlobal(IntPtr hglobal)
   at System.Drawing.Drawing2D.LinearGradientBrush._SetBlend(Blend blend)
   at LaserGRBL.UserControls.ColorProgressBar.DrawBar(Graphics G, Int32 W, Int32 H, Double V, Color C) in E:\LaserGRBL\LaserGRBL\UserControls\ColorProgressBar.cs:line 489
   at LaserGRBL.UserControls.DoubleProgressBar.DrawProgres(Graphics g) in E:\LaserGRBL\LaserGRBL\UserControls\DoubleProgressBar.cs:line 135
   at LaserGRBL.UserControls.ColorProgressBar.OnPaint(PaintEventArgs e) in E:\LaserGRBL\LaserGRBL\UserControls\ColorProgressBar.cs:line 308
   at System.Windows.Forms.Control.PaintWithErrorHandling(PaintEventArgs e, Int16 layer)
   at System.Windows.Forms.Control.WmPaint(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.UserControl.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

lasermyfaceoff, 2021-01-07:
This is the crash error information when I disconnect the laser via the button in the top left. It gives this error everytime I disconnect and closed LaserGRBL. 

Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	LaserGRBL.exe
  Application Version:	4.0.2.0
  Application Timestamp:	5ff444a3
  Fault Module Name:	StackHash_f2cc
  Fault Module Version:	6.1.7601.24384
  Fault Module Timestamp:	5c6e245d
  Exception Code:	c0000374
  Exception Offset:	00000000000bf302
  OS Version:	6.1.7601.2.1.0.768.3
  Locale ID:	1033
  Additional Information 1:	f2cc
  Additional Information 2:	f2cc60f8f16c1ff4741b0bd90b77e806
  Additional Information 3:	f2cc
  Additional Information 4:	f2cc60f8f16c1ff4741b0bd90b77e806

lasermyfaceoff, 2021-01-08:
The issue ended up being the USB cable provided by Ortur....not sure if it was just too high impedance or poor connection but it would trigger the Ortur laser safety function to shut off if disconnected....Thanks for all the assistance. A new and shorter USB cable fixed the problem.

arkypita, 2021-01-08:
I am going to forward this report to ortur, as a reference for their diagnostics. Keep in mind that instead of a problem with the Ortur cable (I have tried several without having errors) it could also be a defect in the socket of the USB port on your PC, which you may not notice with the cable you changed.

closed 2021-01-08 by arkypita