arkypita/LaserGRBL #1255: Exception if using the INFO button when setting line counts - the problem still exists

fuseoppl, 2021-01-09:
Steps to reproduce the behavior:
1. Open an bmp image
2. Use the small blue INFO button to calculate the exact new line count.

My Grbl Configuration:
X-axis travel resolution = **3200**
Y-axis travel resolution = **3200**

_See error:_

LaserGrbl v4.0.4.0
Grbl v1.1h
Wrapper: UsbSerial
Win 10, Professional, 10.0.19042.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.4300.0 built by: NET48REL1LAST_C

TypeOf exception  [System.ArgumentOutOfRangeException]
Exception message [Value of '**3200,000**' is not valid for 'Value'. 'Value' should be between 'Minimum' and 'Maximum'.
Parameter name: Value]
Exception source  [System.Windows.Forms], thread [Main Thread]
Exception method  [Void set_Value(System.Decimal)]

  ----------- stack trace -----------
   at System.Windows.Forms.NumericUpDown.set_Value(Decimal value)
   at LaserGRBL.RasterConverter.ResolutionHelperForm.CreateAndShowDialog(Form parent, GrblCore Core, Double oldval) in E:\LaserGRBL\LaserGRBL\RasterConverter\ResolutionHelperForm.cs:line 44
   at LaserGRBL.RasterConverter.RasterToLaserForm.BtnQualityInfo_Click(Object sender, EventArgs e) in E:\LaserGRBL\LaserGRBL\RasterConverter\RasterToLaserForm.cs:line 697
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.UserControl.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

arkypita, 2021-01-12:
You have a very high resolution. Are you sure that your resolution is ok?

arkypita, 2021-01-12:
I have increased maximum allowed number to 10'000 but in my opinion 3200 is an insane value and could make issue with grbl firmware.

closed 2021-01-12 by arkypita

fuseoppl, 2021-01-12:
I checked various settings and at the maximum resolution of my stepper motor driver during the travel there are the smallest vibrations and the shift works smoothly and quietly, only for these reasons I left these settings. Of course, I do not need such a high resolution, but believe me, with the resolution of 3200, the mechanism works very quietly and smoothly without vibrations.

commit a5fb00c4 (2021-01-12): Fix #1255