arkypita/LaserGRBL #1257: Problem with LaserGrbl connecting to CNC machine

Envirowood, 2021-01-10:
Unable to connect to CNC machine. 
When trying to install CH340 Driver I receive the following error 
 - Windows version 10
 - CNC 3018 pro
 - LaserGRBL version 3.8.7

This is the error message I receive
LaserGrbl v3.8.7.0
Grbl v1.1f
Wrapper: UsbSerial
Win 10, Professional, 10.0.18363.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v2.0.50727\mscorlib.dll 2.0.50727.9151 (WinRelRS6.050727-9100)

TypeOf exception  [System.ComponentModel.Win32Exception]
Exception message [The system cannot find the file specified]
Exception source  [System], thread [Main Thread]
Exception method  [Boolean StartWithShellExecuteEx(System.Diagnostics.ProcessStartInfo)]

   ----------- stack trace -----------
   at System.Diagnostics.Process.StartWithShellExecuteEx(ProcessStartInfo startInfo)
   at System.Diagnostics.Process.Start(ProcessStartInfo startInfo)
   at LaserGRBL.MainForm.installCH340DriverToolStripMenuItem_Click(Object sender, EventArgs e) in E:\LaserGRBL\LaserGRBL\MainForm.cs:line 634
   at System.Windows.Forms.ToolStripItem.RaiseEvent(Object key, EventArgs e)
   at System.Windows.Forms.ToolStripMenuItem.OnClick(EventArgs e)
   at System.Windows.Forms.ToolStripItem.HandleClick(EventArgs e)
   at System.Windows.Forms.ToolStripItem.HandleMouseUp(MouseEventArgs e)
   at System.Windows.Forms.ToolStrip.OnMouseUp(MouseEventArgs mea)
   at System.Windows.Forms.ToolStripDropDown.OnMouseUp(MouseEventArgs mea)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ToolStrip.WndProc(Message& m)
   at System.Windows.Forms.ToolStripDropDown.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

arkypita, 2021-01-10:
You are the second user that is reporting this issue, however i triple checked and LaserGRBL installer is copying the ch340 driver correctly inside lasergrbl folder. Can you check what do you have in this folder?

[image]

Envirowood, 2021-01-11:
I have exactly that in the folder. I have exactly the same issue when I use a different laptop so I believe it is a port issue on the CNC board that needs replacing

arkypita, 2021-01-12:
If you cannot install driver from LaserGRBL you can google "CH340 driver" and download the package from web.

closed 2021-01-16 by arkypita