arkypita/LaserGRBL #1259: Unable to import an image

brittyw, 2021-01-10:
**Describe the bug**
I received the Ortur Laser Master 2 for Christmas and am having a hard time importing the svg I have created.

**To Reproduce**
Steps to reproduce the behavior:
1. File
2. Open File
3. Selected the .svg
4. Target Image box opens
5. Create!
Nothing appears... :( 

**Expected behavior**
I thought once I loaded the image it would appear in the main area of the GRBL software.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware and software configuration (please complete the following information):**
 - Windows version: 10
 - Engraver brand and model: Ortur Lasrt Master 2
 - LaserGRBL version: 3.8.6

arkypita, 2021-01-10:
SVG is a standard with many different implementation, depending on how the file was created, and which program generated it, lasegrbl may not understand it.

usually I open the file in inkscape (it's a free program), I re-save it, and everything works. If it comes out wrong position, I cut it and I copy it in a new file.

If there are any text, you have to turn them into paths. If there are filled parts, you need to take care of the filling in your vector editor. For inkscape there are plugins like "eggbot"

brittyw, 2021-01-10:
I actually did create it in Inkscape, and it is text.  But I'm going to try turning it into a path like you said, thanks!

arkypita, 2021-01-10:
you are welcome, next time please use "discussion button" because this is not an issue of lasergrbl

closed 2021-01-16 by arkypita