arkypita/LaserGRBL #1266: Vettorizzazione non perfetta

sylunder, 2021-01-13:
Ho scaricato l'ultima versione di LaserGrbl e posseggo Ortur Laser Master 2.
Da un po' di tempo la vettorizzazione non risulta perfetta, infatti quando fa l'ultimo passaggio sui contorni risultano sfalsati.
Sbaglio qualcosa? Grazie
[image]

arkypita, 2021-01-13:
http://lasergrbl.com/faq/

arkypita, 2021-01-13:
Hai lo stesso problema anche se usi la modalità vecchia? 

[image]

Comunque leggi le faq e poi ne riparliamo.

closed 2021-01-13 by arkypita

sylunder, 2021-01-13:
Ho scaricato la versione 4.0.5. Ho modificato $32 = 1. Non so se dipenda da quella ma mi sembra meglio. Grazie

arkypita, 2021-01-13:
Sicuramente $32=1 da dei risultati migliori, ma non dovrebbe influire sul posizionamento.
Io guarderei più a problemi meccanici

[image]

sylunder, 2021-01-14:
Ti ringrazio per aver proseguito nel problema... non mi pare ci siano problemi meccanici (l'ortur è nuovo) 

Il disallineamento lo fa solo nell'ultima passata quando ricalca i bordi.

Grazie se mi vuoi rispondere

> 
> 
> 
> 
>     Sicuramente $32=1 da dei risultati migliori, ma non dovrebbe influire sul posizionamento.
>     Io guarderei più a problemi meccanici
> 
>     [image] https://user-images.githubusercontent.com/8782035/104457299-26e0cc80-55aa-11eb-9320-c00ab9767fab.png

_______________
Silvano Sottocasa
335 6315207
silvano.sottocasa@pec.it
sottocas@libero.it

arkypita, 2021-01-14:
Avvita la tua macchina al pavimento e vedrai che non succede più.

Scherzi a parte: LaserGRBL invia alla macchina dei comandi di posizione, i quali sono calcolati con precisione assoluta.

Se tra le linee di riempimento e quella di contorno **che sono inviate con le stesse identiche coordinate**, c'è una differenza, è perché la macchina ha perso la posizione durante la lavorazione.

Può essere perché la macchina si è mossa tra il riempimento e il contorno, perché ha accumulato dei piccoli errori nei movimenti dovuti alle cinghie allentate (la macchina sarà anche nuova, ma l'hai montata e regolata bene? Tutte le cinghie sono tirate correttamente? L'asse è in quadro?), può essere perché il laser si muoveva troppo velocemente e questo ha fatto muovere la macchina etc..

Ripeto, il problema non è di LaserGRBL.

sylunder, 2021-01-15:
GRazie ridurrò la velocità allora perchè la macchina è fermata su una base. le cinghie sono giuste

> Avvita la tua macchina al pavimento e vedrai che non succede più.
>
>     Scherzi a parte: LaserGRBL invia alla macchina dei comandi di posizione, i quali sono calcolati con precisione assoluta. [...]

_______________
Silvano Sottocasa
335 6315207
silvano.sottocasa@pec.it
sottocas@libero.it