arkypita/LaserGRBL #1270: Disconnect usb exception message

StyrosJCG, 2021-01-16:
LaserGrbl v3.8.7.0
Grbl v1.1h
Wrapper: UsbSerial
Win 7, Professional, 6.1.7601.65536, Service Pack 1 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v2.0.50727\mscorlib.dll 2.0.50727.8806 (QFE.050727-8800)

TypeOf exception  [System.ObjectDisposedException]
Exception message [Safe handle has been closed]
Exception source  [mscorlib], thread []
Exception method  [IntPtr SafeHandleC2NHelper(System.Object, IntPtr)]

   ----------- stack trace -----------
   at System.StubHelpers.StubHelpers.SafeHandleC2NHelper(Object pThis, IntPtr pCleanupWorkList)
   at Microsoft.Win32.UnsafeNativeMethods.GetOverlappedResult(SafeFileHandle hFile, NativeOverlapped* lpOverlapped, Int32& lpNumberOfBytesTransferred, Boolean bWait)
   at System.IO.Ports.SerialStream.EventLoopRunner.WaitForCommEvent()
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart()

arkypita, 2021-01-18:
It was reported multiple time.
https://github.com/arkypita/LaserGRBL/issues/990
https://github.com/arkypita/LaserGRBL/issues/1028

It appears to have been caused by the .NET framework system serial driver, not directly by LaserGRBL. Unfortunately, since it is external code, it is difficult for me to catch this error.

I have already received a couple of reports, which seemed to limit the problem to PCs with Windows 7, but  some on w10 say the opposite.

I would really like to solve it, but the biggest difficulty is that on my PC I can't make it happen. Can you tell me if the error is sporadic, or does it always happen? Is there a particular procedure with which you can make it happen?

In the latest v4.0.5 I write a new SerialUSB wrapper to comunicate with the engraver, that replace microsoft implementation. 

If you want you can test by download from here: https://github.com/arkypita/LaserGRBL/releases/download/v4.0.5/install.exe and enable UsbSerial2 connection protocol (before connect!).

[image]

arkypita, 2021-01-23:
Does it solve or not?
Can you try update Microsoft .NET framework? https://www.google.com/search?q=microsoft+.net+framework

Let me know

danielostCZ, 2021-03-28:
Hi, I tried on 2 PCs, the first Win 7 ultimete, the second Win 7 prof. GRBL version 4.2.1. set USBSerial2 (tested even after connection restart) without success. Streaming "RepeatOnError", Threading "Slow". MS NET Framework 4.7. It prints "Unexpected disconnect" during engraving and sometimes while moving. "The port does not exist" appears in the log. I will be happy for any advice, if you need to try something, I am available. The source has 60W (12V, 5A), laser power consumption 15W (4.5W optical power), USB cable with ferrite filter. Grounding is measured and OK. Thank you

closed 2021-06-21 by arkypita