arkypita/LaserGRBL #1272: LaserGRBL not connecting

MekoZeko, 2021-01-16:
Hi. I used my laser engraver 2-3 days ago but today LaserGRBL didn't connecting to engraver. I tried to reinstall CH340 drivers and update LaserGRBL but they didn't work either. LaserGRBL session log : 

17.01.2021 01:03:36.461	Program     	------- LaserGRBL v4.0.5 [64bit] START -------
17.01.2021 01:03:39.737	SetStatus   	Machine status [Connecting]
17.01.2021 01:03:39.740	OpenCom     	Open COM4 @ 115200 baud  (UsbSerial)
17.01.2021 01:03:39.743	SendImmediate	Send Immediate Command [0x18]
17.01.2021 01:03:49.739	OpenCom     	Connection timeout!
17.01.2021 01:03:49.739	CloseCom    	Close COM4 [CORE]
17.01.2021 01:03:49.740	SetStatus   	Machine status [Disconnected]

CH340 Driver Settings : 
[image]

Baud rate seems wrong but i didn't change anything before and it works like that. My machine is Atomstack A5.

arkypita, 2021-01-16:
Baud rate and settings inside COM configuration does not matter.
Just set 115200 in LaserGRBL

arkypita, 2021-01-16:
[image]

try setting hard and soft reset at connect

arkypita, 2021-01-16:
Record an extended log (menu ? - activate extended log) and see if the engraver transmit anything

MekoZeko, 2021-01-16:
[comlog.txt](https://github.com/arkypita/LaserGRBL/files/5825208/comlog.txt)

arkypita, 2021-01-16:
Usually comlog look like this:

00000    00119782    log    Recording session started @ 12/3/2019 12:14:20 AM
00001    84893492    com    Open COM21 @ 115200 baud 
00002    84893834    tx    [3F]
00003    84893936    com    Close COM21 [CORE]
00004    84896505    com    Open COM21 @ 115200 baud 
00005    84896540    com    Open COM2 @ 115200 baud 
00006    84900283    com    Open COM21 @ 115200 baud 
00007    84900371    tx    [3F]
00008    84900410    rx    <Idle|WPos:-116.619,-101.500,0.000|Bf:15,128|FS:0,0|Ov:100,100,100>\r
00009    84901391    tx    [3F]
00010    84901420    rx    <Idle|WPos:-116.619,-101.500,0.000|Bf:15,128|FS:0,0>\r
00011    84902384    tx    [3F]
00012    84902413    rx    <Idle|WPos:-116.619,-101.500,0.000|Bf:15,128|FS:0,0>\r
00013    84903391    tx    [3F]
00014    84903419    rx    <Idle|WPos:-116.619,-101.500,0.000|Bf:15,128|FS:0,0>\r
00015    84904382    tx    [3F]
00016    84904411    rx    <Idle|WPos:-116.619,-101.500,0.000|Bf:15,128|FS:0,0>\r

If you see only tx and no rx it mean that the engraver board is not responding (died or bad firmware)

arkypita, 2021-01-16:
I see your CH340 on COM4 but the log you upload talk about COM3

MekoZeko, 2021-01-16:
I just change the port just to try its COM3 on device manager too

MekoZeko, 2021-01-16:
Should I flash firmware?

arkypita, 2021-01-16:
> Should I flash firmware?

If the board is not dead it could save you.

arkypita, 2021-01-16:
Did you have any other computer to test with?

MekoZeko, 2021-01-16:
I flashed the firmware to v1.1h original in LaserGRBL and it worked. Thanks for your help.

arkypita, 2021-01-16:
you are welcome. I suggest to report your issue to atomstack, just for case history

closed 2021-01-16 by arkypita