arkypita/LaserGRBL #1280: Dithering engraving. Laser stuck along x axis for no reason and resume next to where it should be

Beewap, 2021-01-16:
Today I try dithering engraving for a picture. The result was nice but I don't know why the head had a problem. 
The head should goes from the left to the right on the x axis and goes back, move after on the y axis and do the same left/right and so on... but I had the issue. For some reasons it didn't goes back where it should before moving on the y axis and the laser resume the engraving where it was. So the picture was splited and shifted.

Imagine having a picture of 5x5cm, the laser start and it's ok and after few secondes/minutes (it depends of the picture I think) the head is blocked and don't move (there is a nasty noise when it try to move). The laser resume the picture at this exact point and now the picture  is split and shifted and the final dimension is something 10x5cm. 
I had this problem several times on the same picture and the picture move  again and again and again (it end at something like 15x5cm but I stop the engraving..

[image]

In this picture I try to do a circle and you can see that the upper part move on the left 3 times.

I use LaserGRBL 4.0.5 (lasted version)
Firmware 1.1f
Mostics cnc with 5.5w laser

Any idea ?

arkypita, 2021-01-17:
http://lasergrbl.com/faq/

Beewap, 2021-01-17:
Hi, 

Yes, someone speak about $120-121 and 122, where I had a big value in $120. I'm testing with a lower value and it look pretty good, no issue atm.

Thanks. ;)

closed 2021-01-23 by arkypita