arkypita/LaserGRBL #1294: Problem with new Laser GBRL

Mycki17, 2021-01-20:
LaserGrbl v4.0.5.0
Grbl v1.1f
Wrapper: UsbSerial
Win 10, Professional, 10.0.19041.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll 4.8.4300.0 built by: NET48REL1LAST_C

TypeOf exception  [System.ArgumentException]
Exception message [Ungültiger Parameter.]
Exception source  [System.Drawing], thread [Main Thread]
Exception method  [Int32 get_Width()]

   ----------- stack trace -----------
   bei System.Drawing.Image.get_Width()
   bei System.Drawing.Image.get_Size()
   bei LaserGRBL.RasterConverter.ImageProcessor.ResizeRecalc() in E:\LaserGRBL\LaserGRBL\RasterConverter\ImageProcessor.cs:Zeile 407.
   bei LaserGRBL.RasterConverter.RasterToLaserForm.PbConverted_Resize(Object sender, EventArgs e) in E:\LaserGRBL\LaserGRBL\RasterConverter\RasterToLaserForm.cs:Zeile 688.
   bei System.Windows.Forms.Control.OnResize(EventArgs e)
   bei System.Windows.Forms.PictureBox.OnResize(EventArgs e)
   bei System.Windows.Forms.Control.OnSizeChanged(EventArgs e)
   bei System.Windows.Forms.Control.UpdateBounds(Int32 x, Int32 y, Int32 width, Int32 height, Int32 clientWidth, Int32 clientHeight)
   bei System.Windows.Forms.Control.UpdateBounds()
   bei System.Windows.Forms.Control.WmWindowPosChanged(Message& m)
   bei System.Windows.Forms.Control.WndProc(Message& m)
   bei System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

arkypita, 2021-01-20:
Thanks. It was already reported (https://github.com/arkypita/LaserGRBL/issues/1288) but I can't replicate

In order to solve this problem you should help me to understand in what conditions it presents itself.
What are you doing - exactly - when the error message came out? Can you replicate it or is it sporadic?

arkypita, 2021-01-23:
> Hi, i started the update just today and did some Vector burning on some woodplate.
> After the converting of the picture to the G-Code there is just the message "Oops something get wrong" but i was able to carry out the plot after pushing continue.
> I will observe this in the next time.

Thanks for the info.

I have done some small changes in the code of raster import form, but i don't understand if does it have something to do with it, or if it's the fact that I switched from using microsoft .net framework 3.5 to 4.0

Unfortunately I can't get this to happen on my computer so it's hard to figure out if and how to fix it.

closed 2021-01-23 by arkypita