arkypita/LaserGRBL #1314: English text only

pebcz, 2021-01-28:
"* Options" text is in English only for first 2 types - may be it's defined "hard" in RasterToLaserForm.cs, here?
GbLineToLineOptions.Text = RbLineToLineTracing.Checked ? "Line To Line Options" : "Dithering Options";
[image]

arkypita, 2021-01-29:
Fixed https://github.com/arkypita/LaserGRBL/commit/6218e54068a9b90cd0780530e709cd3472d8e31c

There may be others like this one.

pebcz, 2021-01-29:
Question: you add new items (LineToLineOptions, DitheringOptions) to Strings.resx, but necessary items are also in RasterToLaserForm.resx (GbLineToLineOptions.Text - I think it's not used in any place in source code) - is it correct?

arkypita, 2021-01-29:
> Question: you add new items (LineToLineOptions, DitheringOptions) to Strings.resx, but necessary items are also in RasterToLaserForm.resx (GbLineToLineOptions.Text - I think it's not used in any place in source code) - is it correct?

When you put a control on the form, text label, groupbox, checkbox or other, visual studio create a localization entry automatically in the form resx. It is there and you cannot avoid having it.

But if you want to put a dynamic text in this control, for example that shows a state that changes according to the conditions, then you have to manage the translation manually. 

The original writing remains in the translation file, but in practice apart from the first very short initial drawing, it will never be displayed because the code will immediately set the one corresponding to the status.

pebcz, 2021-01-30:
newly found:

[image]
[image]