arkypita/LaserGRBL #1315: V4.1.0 UPDATE

jshack54, 2021-01-29:
I recently updated to v4.1.0 since I installed update. I ran 2-5 little projects. No problems. Last night I ran 1 and when I started second one. The thing went nuts. While it was running all of a sudden X started making noise and running all over the place.
Example tried to home and instead of going to normal zero it went about half way and changed direction. When to jog mode and trying to jog right it went left then back a couple of mm and stop then went right. I have done a hard reset unplugged and plug back in power and usb with no results.   I am not very technical with steppers switches, wiring and configuring

arkypita, 2021-01-29:
I wouldn't say it's a problem with LaserGRBL but with your engraver.

jshack54, 2021-01-29:
That's what I thought. I changed stepper and adjusted the belt. I thought motor might be bad. Same result with different motor.

arkypita, 2021-01-29:
Check the cables. Remember that you must NEVER attach / detach cables, drivers, and motors with the machine powered on. The motor driver burns out.

Check your grbl configuration. I have seen some cases where it happened that the machine had a corrupt configuration.

Flash the firmware again.

jshack54, 2021-01-29:
I changed stepper motor and adjusted belt. No change just wondering if I may have a stripped belt gear there.
It will work and then all of a sudden the laser beam starts vibrating. And then it bounces. The sound it makes while running is almost like if you run past the stop. example grinding gears.

arkypita, 2021-01-29:
Are you using high speeds? are you using microstepping? 
From the description it seems the behavior that the engines do when you try to make them go too fast.

What are your config parameters $0, $1 and $100 to $122

jshack54, 2021-01-29:
[image]

jshack54, 2021-01-29:
I will try to send a video when I get home today.

arkypita, 2021-01-29:
Values might also be fine. Perhaps the maximum speeds (110,111, 112) are a bit high, I would put them at 6000.

What kind of engraver is it?
Did you buy it as an assembly kit or did you design it yourself?
Which control board it use?
Are these values the original ones, or have you changed them recently?

jshack54, 2021-01-29:
[image]
The settings have not been changed. The only settings I change are when you open an image ie speed size from when you open file to run.

jshack54, 2021-01-29:
I cant get video to send.

peter9811, 2021-02-11:
Maybe try deleting the configs and do a reset of it 
I used this to get it: https://ultimatecnc.softgon.net/en/download

Ozzybanan, 2021-02-21:
Reinstall the program. Make a Grbl update. Check if it's OK. In the settings you have $ 3 = 5, it means you have reversed direction of X axis and Z axis. Send $ 3 = 0 to machine and check. Write if it's OK.

njmartinez1234, 2021-03-29:
Hi, first i want to say thanks for doing this software and for take the effort  and time to support it.

Im having the same issue, I've read comments and faq section and all documentary i could find, but in my case, the problem starts when grbl shows this error "Bad number format"... the machine gets crazy and starts burning lines everywhere, i have to unplug it, beacuse it's the only way to stop it.. I think it migth be problem with the main board, but maybe you can help me to confirm my theory (i've tested serveral modes and engraving speeds, but the issue remains).. i've even use the Emulator mode, and the  programs finish the engrave process.. i've disconnect the step motors in order to see if it may finish, but nothing seems to work.

I've uploaded some videos and images to google drive,  if someone wants to check
[https://drive.google.com/drive/u/0/folders/13A0v-LeHwY90nVmK7gmiSnUKjJo-_zyQ](url)

 Thanks for your support

arkypita, 2021-03-29:
> Im having the same issue, I've read comments and faq section and all documentary i could find, but in my case, the problem starts when grbl shows this error "Bad number format"... the machine gets [...]

In the video I see you are using one of those machines that have an "offline controller", that box with the display. As far as I am aware, based on the feedback given to me by several users, that controller tries to communicate itself with the machine at the same time as LaserGRBL, creating confusion in the protocol.

You should try to figure out if it is possible to disable this offline controller.

arkypita, 2021-03-29:
I have update [LaserGRBL faq](https://lasergrbl.com/faq/) with your issue.

njmartinez1234, 2021-03-29:
Ok, I'll contact to twotrees's support to figure out how to do what you suggested. I'll write back if i get any solution.

Thank you!

closed 2021-06-21 by arkypita