arkypita/LaserGRBL #1322: it goes beyond the supposed coordinates

alestanga, 2021-02-01:
when I press the start button it goes beyond the supposed coordinates

let's put it should to go to X100-Y100 and then turn on the laser and move:
when I press start, it goes to X100-Y100, turns on the laser, and then it continues to move on the X and Y axis to the infinite and beyond with the laser turned on.
then I press stop, it asks me to restart, if I say yes it goes to X100-Y100 and goes on with the program as it should.
I can provide you with a video of the behavior.

the controller is a SainSmart Genmitsu 3018

alestanga, 2021-02-15:
I've found the issue.
it was a configuration problem.

closed 2021-02-15 by alestanga

arkypita, 2021-02-16:
if you wish you could explain what the cause was and how you solved it, so that it helps others who may have the same situation

alestanga, 2021-02-22:
I had a G91 right after G90.
removing G91 fixed the issue:
[image]