arkypita/LaserGRBL #1323: Adobe Ai SVG is scaled down in LaserGRBL

Orangbelanda, 2021-02-01:
The vector graphics in LaserGRBL is about 20% smaller than the original SVG file from Adobe Illustrator. No idea why this happens, but it seems conssitent.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware and software configuration (please complete the following information):**
 - Windows version
 - Engraver brand and model
 - Grbl version (i.e. 0.9j, 1.1f)
 - LaserGRBL version

**Additional context**
Add any other context about the problem here.

Orangbelanda, 2021-02-02:
[100x100onA4.zip](https://github.com/arkypita/LaserGRBL/files/5909656/100x100onA4.zip) Th Ai and SVG files as example.

arkypita, 2021-02-02:
If I open it in Inkscape it show 75*75 rectangle in a non A4 page size

[image]

[image]

svenhb, 2021-02-02:
As I know: the standard is 96 dpi (used in inkscape) but AI  may save with 72 dpi.
https://community.adobe.com/t5/illustrator/illustrator-svg-export-dimension-problem/td-p/10159751

Orangbelanda, 2021-02-02:
Yes ! This must be the cause. If I scale up the Ai file to 133,333 percent, the dimensions saved to SVG are exactly right. Its a bit weird  work around, but it works :-) Would be nice if LaserGBL could do this for me (while import Ai SVG).

svenhb, 2021-02-02:
I would say it is a problem of AI, since 96 dpi is the standard :-)

Orangbelanda, 2021-02-02:
OK, OK, OK, I am not devending Ai ;-)

arkypita, 2021-02-02:
SVG support in LaserGRBL is in a rather embryonic state. The main problem is that there is no single standard of SVG, and that there are dozens of possible transformation / scaling operations, units of measurement, viewports etc, which different programs implement in different ways, which is a headache to manage them all.

Moreover, it often happens that if you touch something for one then it no longer works for another.

The other problem is that the code I use to read the SVG I didn't write it myself, but I got it from another third party project, so I don't have 100% knowledge of what it does, and how to change it.

It is for these reasons that SVG support is so poor at present in LaserGRBL.

In my tests I preferred to align myself with the SVG of inkscape, as being a free editor it can be used by everyone.

Orangbelanda, 2021-02-03:
I understand such headaches, not as a programmer, but as an industrial designer. And I appreciate your choice for Inkscape compatibility. How about EPS file format, will that give such headaches as well ?

Orangbelanda, 2021-02-03:
Strange, I tried an other 133,333 percent scaled SVG file from Ai but the resulting scale in LaserGRBL was wrong. Anyway, I decided to use Inkscape to open the original Ai file, and than save it to SVG (Inkscape version), which works very well.

sanyats, 2021-03-03:
у меня при работе с SVG файлами LaserGRBL уходит в критическую ошибку, что может быть, как исправить?

Orangbelanda, 2021-03-04:
Файл SVG должен создаваться Inkscape, поскольку разные программы создают разные версии SVG. Хочешь верь, хочешь нет ;-)

closed 2021-06-21 by arkypita