arkypita/LaserGRBL #1340: Displays a large number of program source errors when running by Visual Studio

ashkan4818, 2021-02-15:
Hi, I get a lot of errors when I run the source file software in Visual Studio. How can I solve this problem?
[image]

[image]

arkypita, 2021-02-15:
Look like your VS installation doesn't have C# 7.0 support

[image]

ashkan4818, 2021-02-15:
The problem was solved. Thanks for your tips

PeterBrewmaster, 2021-02-24:
I have the same problem. Tried it on several different computers using VS 2015 and VS 2019. In VS 2019 the C# language version is 9.

arkypita, 2021-02-25:
I don't know how to help you!

I use VS2017 Community Edition, default installation.
All the source file of LaserGRBL is included in the github repository, no dependency of third part library.

PeterBrewmaster, 2021-02-25:
I installed VS 2017 - same problem. Even when setting the language version to 7.0 (from preset 7.3). It would be great to hear what ashkan4818 did to solve it.

ashkan4818, 2021-02-27:
hi @PeterBrewmaster  i installed VS2017 Enterprise and required components . You must be sure to unblock the LaserGrbl zip file before extract the zip file. to do this : right click on LaserGrbl zip file and go to properties and click on "unblock". 

That's all I did and my problem was solved.
I hope you find it useful

PeterBrewmaster, 2021-02-27:
> hi @PeterBrewmaster i installed VS2017 Enterprise and required components . You must be sure to unblock the LaserGrbl zip file before extract the zip file. to do this : right click on LaserGrbl zip [...]

Thanks. I tried that but all the errors were still there. BUT what I did, I run/started the program and it worked! When stopping it there were no errors left. Must be some issue with Visual Studio 2015, 2017 and 2019.

arkypita, 2021-02-27:
So you compile the code.

PeterBrewmaster, 2021-02-27:
Yes.

ashkan4818, 2021-02-27:
@PeterBrewmaster  was your problem solved?

PeterBrewmaster, 2021-02-28:
> @PeterBrewmaster was your problem solved?

I think so. I have not tried it with the engraving machine yet though. I am building a machine to label cartons with engraving. So I need to be able to start engraving and be noticed when engraving is completed for each carton, so that this machine can bring next carton to be engraved. The LaserGRBL software is really amazing.

closed 2021-03-23 by arkypita