arkypita/LaserGRBL #1349: Help to solve the problem in developing LaserGRBL software

ashkan4818, 2021-02-21:
Hi, I'm trying to add some features to your LaserGRBL software . I created a user control and did my coding in it. Now I have to send a series of G-codes to the CNC machine using a button, but when I press the button, I get the following error. Can you help me solve this problem? I appreciate and thank you for any of your guidance. 
[image]

arkypita, 2021-02-22:
This photograph is not enough to allow me to help you. It's like you send a picture of your engine to the mechanic asking him why the car won't start.

ashkan4818, 2021-02-22:
:-D , OK Please tell me what information you need, so that I can send it to you
. I believe in your programming skills

arkypita, 2021-02-24:
Share the code of the project or at least the class you are using.
Null reference means that you have not initialized / assigned "Core" variable value.

ashkan4818, 2021-02-27:
dear @arkypita i was able to solve this problem, too. Thank you

closed 2021-03-23 by arkypita