arkypita/LaserGRBL #1362: Freezes and stops in work constantly occur.

serzhants, 2021-03-12:
During operation, at any time there is a stop and freeze, which is eliminated by pressing the Reset GRBL button and restarting the program from the point of stopping. After a while, the operation stops again. And so it is constantly.

At the moment of stop, the sound of connecting the device sounds. One gets the impression that the device is reconnecting.
In the system, the device on the COM port is displayed constantly.

I am attaching a screenshot of the program during such a stop and an extended log.

P.S.: Sorry, english translated by google. =)

[image]

[comlog4.txt](https://github.com/arkypita/LaserGRBL/files/6130323/comlog4.txt)

Hardware and software configuration:
 - Windows 10 Pro x64
 - LaserGRBL version 4.2.1
 - Engraver based by Arduino Uno R3 (CH340G) ATmega328 + CNC Shield V3
 - Grbl version: v1.1h original (25.08.2020) (install wiht LaserGRBL ->Tools -> Flash GRBL Firmware -> v1.1h original (25.08.2020)

arkypita, 2021-03-12:
Dear @serzhants 
Thanks for the detailed and well documented report.

Yours seems like a classic case of communication issue due to electrical noise.
It is possible to see, in command log, some incomplete lines sent by the engraver (the pink ones) that LaserGRBL cannot recognize.

Try to send the gcode in check mode. Type "$C" in "type gcode here" box before sending gcode, LaserGRBL stream the whole file to the engraver, and the engraver reply to communication, but the motor are not moved, so there is no electrical noise (by the motor).

If this test pass, you have understand your problem.

Please note that CH340G is know for random communication issue. An original Arduino Uno is usually more reliable.
Read https://lasergrbl.com/faq/ for more detail about your issue and possible solution.

serzhants, 2021-03-12:
I tried to run the gcode in test mode using "$C", and both times the test failed, and it hung.

I am using a well-shielded USB cable.

I am attaching extended logs of both attempts.
[comlog6.txt](https://github.com/arkypita/LaserGRBL/files/6130872/comlog6.txt)
[comlog7.txt](https://github.com/arkypita/LaserGRBL/files/6130873/comlog7.txt)

arkypita, 2021-03-12:
comlog6, line 00472: the string "Chek" is missing of one char ("Check" is the correct syntax).
comlog6, line 00859: the string "o\r" is missing of one char ("ok\r" is the correct syntax).

I don't have time to look for more of these events, but it's obvious that this reveals that there is some problem with what your arduino board sends to LaserGRBL.

If you have another arduino board try using it, even disconnected from the CNC shield, to see how it behaves.

You may also have the cable connector making bad contact.

I already told you, it's all written in the faq.

serzhants, 2021-03-12:
Thanks for everything what you do.

I kind of figured out what the reason was, in the coming days I will do some tests and complement this issues.

CPT-Murdock, 2021-03-13:
Hi 3dbro,

do you have any information, regarding the Problem? I have the same isse. Gerne auch in Deutsch.

3dbro, 2021-03-14:
Hi, @CPT-Murdock.

Please write what is the complete marking on the microcontroller chip of your arduino?

Here's an example: Atmel Mega328P U-KR 354A3P 1942T6A

CPT-Murdock, 2021-03-16:
Unfortunately there is nothing written... but ir must be a Woodpecker. Could it be a buffer overflow? The Problem occurs only with "insert fill"

StuartB4, 2021-03-16:
@CPT-Murdock. Can you post a picture of the board.

closed 2021-04-16 by arkypita