arkypita/LaserGRBL #1367: vectorization tool contour doesn't consider Linewidth

Ratofl, 2021-03-17:
When you use the vectorization tool the generated contourline is centerd around the original contour of the picture. however, to get the small details right (I use it to draw pcb Layouts) you need to shift the center of the contour line inwards for half the linewidth (for 0.3mm linewidth, which would be 3.33 lines/mm, that would be 0.15mm contourshift. Or for 0.1mm linewidth, it would be 10 lines/mm and 0.05 mm contourshift)

**Hardware and software configuration (please complete the following information):**
 - Windows version
 - Engraver brand and model
 - Grbl version (i.e. 0.9h)
 - LaserGRBL 4.2.1

arkypita, 2021-03-23:
I don't understand. Can you add some pictures?

Ratofl, 2021-03-23:
sure. snapshot of a part of some primitive pcb:
[image]
...vectorization...
[image]
...toolpath...
[image]

...problem...
[image]

I hope this helps. This is simply to demonstrate the problems with the contourline. I'm super content with the inset filling.

closed 2021-04-16 by arkypita