arkypita/LaserGRBL #1375: Telnet/Webservice connection errors

j-lang, 2021-03-20:
**Describe the bug**
After a view seconds of running the communication to the laser machine stuck.

**To Reproduce**
Load file: Circle with 10mm
Start plotting. 
3 possible results:
1. Best: it works
2. Some communication errors are shown, see screenshot: the plotter answers "bad number format" or ""illegal command"
3. The plot stops after some commands.
With USB connected, everything works perfect, but I like to have the engraver outdoors. But the cable length limit is also 5m. 

Also repeating reading the parameters with "$$": after a few commands there is no answer, the next $$ will work.
Unfortunately there is no way the get an error prediction, I can make 3 to 5 successful test-plots and the final plot a minute later on the "good" part has an error an the part is destroyed. 

**Screenshots**
[image]

**Hardware and software configuration (please complete the following information):**
 - Windows 10 desktop PC connected to LAN (20H2, HP Z2 Workstation)
 - EleksLaser-Mana Se Board
 - Grbl 1.1h
 - LaserGRBL v4.2.1 (and older)
 - ESP 8266 with telnet client (same with webservice client) https://github.com/arkypita/ESP8266-SerialTelnet 
 - ESP 8266 adapter board (3,3V to 5V conversion)
 - Arduino IDE 1.8.13 with up to date libraries and board-support

**Additional context**
The same configuration worked on an old PC last summer (I use the laser engraver 3 times a year) but only with "synchronous/slow" setting with a older LaserGRBL version but with the same hardware and firmware.  With "buffered" i had the same problem. After a half year I wanted to engrave some parts and got these errors. The engraver is situated on the balcony with a distance of 5 meters. An additional access point situated 1m distance to the engraver does not solve the problem. For crosschecking I installed the webservice firmware from https://github.com/arkypita/ESP8266-SerialWebSocket.

j-lang, 2021-03-20:
Trivial solution: 
The communications errors where homemade. The communication wires from ESP8266 to the Mana board ware running very close the the ESP antenna. I shortened the wires, turned and moved the ESP adapter board with the ESP an now it works again.

closed 2021-03-20 by j-lang

arkypita, 2021-03-23:
Good to know.

What program does generate the g-code? I don't believe you need so much precision (decimals)

j-lang, 2021-03-25:
https://github.com/JTechPhotonics/J-Tech-Photonics-Laser-Tool/