arkypita/LaserGRBL #1389: Grbl and Ortur keeps stopping midway laser engraving project

DAwoodworking, 2021-03-25:
See pics attached.
the LaserGbrl and Ortur keeps stopping mid print and have to restart the program and wasted several nice wood projects that i was trying to engrave.
We are new to this, so it showed to send the attached pictures to this group to try to unersatnd the issue and what i can do to fix this.
Thank you in advance for your help

[image]
[image]
[image]

arkypita, 2021-03-25:
Hi
LaserGRBL is open software not related to any specific laser engraver hardware or vendors.
I suggest you to contact the manufacturer of your engraver for specific and specialized help.

- [Ortur user group](https://www.facebook.com/groups/orturusergroup) (community)
- [Ortur customer support](https://ortur.tech/support/)

In meanwhile  make sure you have read the [LaserGRBL FAQ](https://lasergrbl.com/faq) where you can find the answers to the most common questions.

arkypita, 2021-03-25:
PS: i guess your PC USB port/connector has issue.

Milan-As, 2021-03-26:
I solved the same problem. Replacement of test cables from XP, W7, W10 different crossings. This resulted in suspicions of HW and SW cutters. Finally, a test of various CH340 drivers helped. For W10 installation CH341ALL.

arkypita, 2021-03-26:
Ortur does not have CH340 chip

leelouric, 2021-04-09:
i fixed the problem in windows to disable usb power saving after 14 hours engraving he still run

arkypita, 2021-04-09:
Hi @leelouric 
I will updata lasergrbl faq with your hint.

Can you put some screenshoot and description of how to configure usb power saving?

leelouric, 2021-04-09:
Here the screenshot shots
Budnon French sorry for that

closed 2021-04-16 by arkypita