arkypita/LaserGRBL #1393: Laser Configurations are not shown.

hasanyucel, 2021-03-28:
[image]

My laser power is weak. Laser configurations are not shown. How can I fix it?

gmmanonymus111, 2021-03-28:
What is your GRBL version? (LaserGRBL should tell you, once you've connectred to your machine)

hasanyucel, 2021-03-28:
Machine: Eleksmaker A3 Pro, LaserGRBL version v4.3.0

gmmanonymus111, 2021-03-28:
I meant this:
[image]

hasanyucel, 2021-03-28:
Ah, sorry. It is 0.9i

[image]

gmmanonymus111, 2021-03-28:
As far as I know, laser mode is not yet implemented in GRBL v0.9, but there should be spindle configuration (aka laser power setting, `30` & `$31`). Maybe your machine needs an updated firmware?

hasanyucel, 2021-03-28:
Thank you. I will search for it.

closed 2021-03-28 by hasanyucel

moonshine76w, 2021-03-28:
I'm not sure if you tried this, but try raising and lowering the actual laser with that little screw on the side, It will raise and lower it. You want it to be at the finest point. Hope this helps.