arkypita/LaserGRBL #1413: Problema, passthrough

Francesco-Rugg, 2021-04-15:
Ho un problema con la mia macchina, praticamente impostando una foto in modalità "passthrough" In fase di lavorazione il laser salta qualche linea, allego due foto, quella con più linee saltate è a 120 dpi mentre quella con meno linee è a 125 DPI, cosa posso fare? 
Ho provato a cambiare velocità, sia lenta che veloce, nessun cambiamento, ho cambiato anche il valore $120-121. Non riesco a capire quale potrebbe essere il problema. 

La mia macchina è una Openbuild con scheda grbl MKS dlc 2.0 settata su un microstepping di 1/16

Allego Foto

120 DPI

[image]

125 DPI

[image]

gmmanonymus111, 2021-04-15:
I've made 2 test with passthrough, but mine has no noticable lines (bolt-driven engraver).
[image]

Upper one is 200 DPI, the darker one 300 DPI.
If your machine is belt-driven, it might be the calculated steps sometimes get rounded, thus leaving empty lines. You could try using higher DPI resolution (300 is printing standard AFAIK), or one that is compatible with your stepping resolution (`$100` & `$101`).

arkypita, 2021-04-15:
All those waves in your engraving make me think that you have improper tensioned belt.

[image]

You can set a resolution that is multiple of your step, by using the resolution helper.

[image]

Francesco-Rugg, 2021-04-15:
@arkypita Hi, thanks you for the fast response. 
With line to line option I don't have the problem of skipping line, but only if I use Passthrough. 

This is the photo with 130 DPI Line to line mode with 5.000 line/mm

[image]

arkypita, 2021-04-15:
What is your $100/101?

Francesco-Rugg, 2021-04-15:
@arkypita 

$100=80.006 (x, step/mm)
$101=80.006 (y, step/mm)

I had calculated this number to fine tuning the machine

Francesco-Rugg, 2021-04-15:
@arkypita 

$100=80.006 (x, step/mm)
$101=80.006 (y, step/mm)

I had calculated this number to fine tuning the machine

EDIT in the image that I posted in the comment before there is a chain that makes the shadow on the photo. 

This is a better one

[image]

arkypita, 2021-04-15:
When you use line2line the original image DPI is not used. Image is scaled up/down to match the output size at the target resolution (as line/mm). The distance between two lines in this case is the inverse of the resolution.

i.e 5.000  -> 1/5 = 0.2mm

0.2 is a reliable resolution, as it is a (least common) multiple of your $100 (80 => 1/80 => 0,0125 and 0.2/16 = 0,0125) your motor make exactly 16 step between one line and next.

When using passthrough the original image DPI is used both for determine distance between two lines and image size.

120 dpi -> 25.4/120 = 0,2116666666666666666666666666667mm

Did you get there by yourself?

Francesco-Rugg, 2021-04-15:
@arkypita yes. 

Now, do I need to test another Control board?

arkypita, 2021-04-15:
no, you need to use a DPI that your engraver is able to respect.
Any DPI that give a line multiple of 0,0125mm

try 127 DPI

Francesco-Rugg, 2021-04-15:
@arkypita ok, I understand. Tomorrow I'll do a test and let you know. But I think this solve the problem, thank you for the help

arkypita, 2021-04-15:
pensa a un righello e pensa che la tua macchina può fare passi solo da 0.9mm
se tu imposti 1 linea/mm il gcode gli dice di fare:  0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10

ma la macchina non può farli fisicamente, quindi andrà nella posizione approssimativamente più vicina. fino ad un certo punto la posizione più vicina sarà il multiplo inferiore di 0.9 ma ad un certo punto la posizione più vicina sarà il multiplo superiore.

in quel momento invece che un salto di 0.9 c'è un salto di 1.8

[image]

closed 2021-04-16 by arkypita