arkypita/LaserGRBL #1429: Very high engraving speed!

Farhankhosravi, 2021-04-20:
Hi
I am trying to engrave with a 500mw laser... as you know, i need an slow engraving speed... so i tried to set Engrave speed to 1 mm/min (shown in picture); but when i start engraving, the speed is same with 1000 mm/min !!
[image]

i mean, my device works very fast at every speed, and changing speed number **won't affect** anything...
I attached my GRBL config down here... but i think it's all ok; and problem is inside LaserGRBL software...
[GRBL Config (LaserGRBL).zip](https://github.com/arkypita/LaserGRBL/files/6343346/GRBL.Config.LaserGRBL.zip)

what is problem?... :(

arkypita, 2021-04-20:
No problem inside lasergrbl.
Problem is in your configuration: acceleration.

read https://lasergrbl.com/faq

arkypita, 2021-04-20:
I cannot say your config is good or wrong because i know nothing about what engraver you have,
However resolution and acceleration seem to be low.

Farhankhosravi, 2021-04-20:
> No problem inside lasergrbl.
> 
> Speed is shown upper-right while engraving (near F). I am sure that if you set 1 you have F1

Thank you for your answer....
yes; the Upper right speed shows 1 (and even 0) speed, but in fact, it is very very higher
you can see this small video... compare speed number with blue Plus indicator speed! it moves very fast..

https://user-images.githubusercontent.com/77070895/115399508-8976e380-a1fd-11eb-9847-21676bc71bfb.mp4

Farhankhosravi, 2021-04-20:
in fact, LaserGRBL wrongly thinks that speed is 1... but it's much higher (why?)...

arkypita, 2021-04-20:
LaserGRBL command speed 1 for linear tracing movement (G1).
You set an override of 1.74x so you have 1-2 of speed.

Rapid movement G0 (move between whitespaces) is always executed at fastest speed by the controller.
Fastest speed is configured by $110/$11

arkypita, 2021-04-20:
Did you read [FAQ](https://lasergrbl.com/faq)?

Farhankhosravi, 2021-04-20:
> LaserGRBL command speed 1 for linear tracing movement (G1).
> You set an override of 1.74x so you have 1-2 of speed.
> 
> Rapid movement G0 (move between whitespaces) is always executed at fastest speed by the controller.
> Fastest speed is configured by $110/$11

Sorry, but i didn't understand.
$110/111 are set to 500. and this is max speed for **G0** or **whitespaces**. it's all right for whitespaces; but in my case, engraving speed is very fast too. i mean, im ok with high speed in whitespace, but want much slower speed on engraving...

now you have my config file... which thing is wrong and must be changed to fix my problem? i have read the FAQ but i found same things there.... and didn't find anyway to decrease just **engraving** speed...

Farhankhosravi, 2021-04-20:
I tried to change every option, and problem didn't solved.
### Engraving speed is always very high and wont decrease...

maybe the problem is because of latest version? is that good idea to try old versions?...

StuartB4, 2021-04-20:
What machine is it, these settings seem to be really low for the steps per mm for the motors. 
$100=13.300 (X-axis travel resolution)
$101=13.300 (Y-axis travel resolution)

These are your settings.
$0=10 (Step pulse time)
$1=25 (Step idle delay)
$2=0 (Step pulse invert)
$3=3 (Step direction invert)
$4=0 (Invert step enable pin)
$5=0 (Invert limit pins)
$6=0 (Invert probe pin)
$10=1 (Status report options)
$11=0.010 (Junction deviation)
$12=0.002 (Arc tolerance)
$13=0 (Report in inches)
$20=0 (Soft limits enable)
$21=0 (Hard limits enable)
$22=0 (Homing cycle enable)
$23=0 (Homing direction invert)
$24=25.000 (Homing locate feed rate)
$25=200.000 (Homing search seek rate)
$26=250 (Homing switch debounce delay)
$27=1.000 (Homing switch pull-off distance)
$30=1000 (Maximum spindle speed)
$31=0 (Minimum spindle speed)
$32=1 (Laser-mode enable)
$100=13.300 (X-axis travel resolution)
$101=13.300 (Y-axis travel resolution)
$102=250.000 (Z-axis travel resolution)
$110=500.000 (X-axis maximum rate)
$111=500.000 (Y-axis maximum rate)
$112=500.000 (Z-axis maximum rate)
$120=10.000 (X-axis acceleration)
$121=10.000 (Y-axis acceleration)
$122=10.000 (Z-axis acceleration)
$130=38.000 (X-axis maximum travel)
$131=38.000 (Y-axis maximum travel)
$132=200.000 (Z-axis maximum travel)

arkypita, 2021-04-20:
> maybe the problem is because of latest version? is that good idea to try old versions?...

No problem inside lasergrbl. (again)
LaserGRBL commands F1, it is clearly visible in the first frames of the video.

[image]

Motor motion is executed by GRBL firmware running into the controller, based on its configuration.
Once LaserGRBL send F1 there is nothing more that concern how LaserGRBL work that can affect motor speed: everything is the responsibility of the controller.

arkypita, 2021-04-20:
It's definitely something that concern your $$ configuration.
I did some tests and with a "normal" configuration I have a totally different behavior.

arkypita, 2021-04-20:
But if I use your configuration it behave exactly as your video.

arkypita, 2021-04-20:
Really really strange... use 13,300 as step/mm ($100, $101) or using 13,000 does a lot of difference.
Try yourself by using 13,000

I think this is a bug of GRBL firmware. Try writing to https://github.com/gnea/grbl

arkypita, 2021-04-20:
> Really really strange... use 13,300 as step/mm ($100, $101) or using 13,000 does a lot of difference.
> Try yourself by using 13,000

Edit: 13 or 13,3 no difference.
Difference is between 13 or 80 (that is a more common value)

As @StuartB4  already say: _"these settings seem to be really low for the steps per mm for the motors."_

Maybe you are hitting a limit of GRBL that does not allow to use low speeds with soo low step/mm
However, this is not a stuff concern LaserGRBL but grbl, so please write here: https://github.com/gnea/grbl

closed 2021-04-20 by arkypita

StuartB4, 2021-04-21:
@arkypita.

Looking at the maximum travel size setting he has in $130 and $131 I think it must be one of those 
engravers that use dvd rom stepper motors. They usually have the same size engraving area.

That's what I made my first one with. But the steps per mm were 53.333.
But it would depend on if he has microstepping option on the motor drivers.

Like this. 

[image]

Farhankhosravi, 2021-04-22:
> @arkypita.
>
> Looking at the maximum travel size setting he has in $130 and $131 I think it must be one of those
> engravers that use dvd rom stepper motors. They usually have the same size engraving [...]

Yes. Exactly.
I found that grbl does wrong when your step/mm is very low... If you have low steps, speed won't match with real movement of machine...
So i increased microstepping to 300, and now grbl speed matches with reality. I think that grbl is not made for low resolution steppers.....

Anyway, thanks to all. Problem solved 🙏🏻

StuartB4, 2021-04-22:
Good to know you sorted it out. 👍