arkypita/LaserGRBL #1434: Low power output at Constant power mode

Farhankhosravi, 2021-04-23:
Hello..
As you know, LaserGRBL has a checkbox (in settings) that turns on/off "support PWM".
when you are disabled that, all Engrave progresses must start with S1000 laser power... lower number may harm laser device.
so as i tried, if you use "Vectorize!" option for importing image, laser will start with **S1000** (picture 1) with 5V output and everything goes fine... BUT if you use "1bit BW Dithering", program starts engrave progress with **S250** and 1.3V output! (picture 2). it makes lot of risks for non-TTL lasers...
Please, take a look on that. (point: i always select "M3 constant power" option)
Sincerely.

**Screenshots**
[image]
[image]

**Hardware and software configuration (please complete the following information):**
 - Windows version: 10
 - Grbl version: 1.1h
 - LaserGRBL version: 4.3.0

alastaira, 2021-04-23:
In the "Target Image" dialog, were you setting S-Max to 1000 in both cases?

gmmanonymus111, 2021-04-23:
+1 for this, but it seems both `Vectorize`, `Centerline`, `Pasthrough` and `1bit BW dithering` produces wrong code. For me, the max `S` value should be 12000, but for `Vectorize` and `Centerline`, it created the code with 3600. This was previously was set as S-Max value when importing image while PWM support was enabled. `Pasthrough` and `1bit BW dithering` is capped at 255.
(↓ shows the settings when `Support PWM` is enabled)
[image]
(↓ shows the the generated code when disabling `Support PWM` then importing the image again)
[image]
(↓ 1Bit dithering code, simply set to 255, but my max is still 12000)
[image]
(↓ `Centerline` is the same as `Vectorize`, using the value previously set when `support PWM` was enabled)
[image]
(↓ `Passthrough` seems to be setting the power to 255)
[image]

@alastaira while `Support PWM` is disabled, you can't set `S-Max` value
[image]

StuartB4, 2021-04-24:
@gmmanonymus111

Your settings must be off. You can't have more than 100%.  100% is full power.
Thats the reason you are getting the red boxes around the  S-MIN and S-MAX boxes. 

You have S-MAX set to 3600 but your $30 number must be lower, probably 1000. 
1000 is 100%, laser full on. So setting it to 3600 is not giving you any more power. 

Also around Engraving speed, Border speed. Which means you are telling it to go faster than the 
speed set in grbl. $110 and $111.

gmmanonymus111, 2021-04-24:
@StuartB4 
My bad with laser power, but that does not change the behaviour. (I forgot to change them before taking screenshots). For border and speed, ignore those, as they are also only for demostration purposes. (It's only emulator, I need it to generate GCode, I don't actually want to engrave it)
[image]
[image]
After changing `Support PWM` to disabled:
[image]

StuartB4, 2021-04-24:
Ah right. Ok. thanks. 👍

Farhankhosravi, 2021-04-24:
Well ... now that the bug has been proven, is there a way to bypass it now? or do we have to wait until the next update is released?

gmmanonymus111, 2021-04-24:
You could try saving the GCode, editing the S-Value with a text editor, save the file, then open the edited GCode in LaserGRBL

arkypita, 2021-04-24:
I agree with @Farhankhosravi that this is a bug.

LaserGRBL is designed for machines that **have** PWM control. The switch "without PWM" is to try to make it usable even on those markers that unfortunately are of poor quality and do not have a driver capable of modulating the laser.

The purpose of that switch is to remove all options (like line2line, but now that I think about it, passthrough and M4 should also be removed) and always use the laser at maximum power by commanding only M3 and M5.

The S command with a non-PWM marker should not be generated, or if generated it should always be equivalent to the max value, so as to turn on the control pin at 5V

Currently, instead, a value of S is commanded which is the last one that was used (compatibly with the mode used).

Thanks for letting me know, I will fix shortly.

Farhankhosravi, 2021-04-26:
> I agree with @Farhankhosravi that this is a bug.
>
> LaserGRBL is designed for machines that **have** PWM control. The switch "without PWM" is to try to make it usable even on those markers that [...]

Thanks a lot.
Also.., two more questions:
1- is there any way to disable "border" engraving in "Vectorize!" Mode, and just applying it's "filling"? This mode is very optimized in comparing with 1bit BW dithering... But bordering makes my picture very thick. (And also, it has a lot of options that 1bit dithering doesn't.)
2- will you make a feature in newer versions, to gray out every engraved line in live view? I think it will be helpful.

arkypita, 2021-04-26:
> 1- is there any way to disable "border" engraving in "Vectorize!" Mode, and just applying it's "filling"?

I am also thinking about this option, because for many it creates problems (when the machine is badly configured or has mechanical defects, in this mode they show up a lot). The purpose, however, is to "clean" the jagged edges left by the filling.

Maybe I could already do it in an upcoming version, I just have to figure out how to congeniate the user interface to be simple.

> 2- will you make a feature in newer versions, to gray out every engraved line in live view?

No.

It is something I have already done some tests on but the evidence is that this puts too much weight on the rendering in terms of resource consumption (CPU + Memory) and other problems for example when you have to restart the work. It requires rewriting-revision of many parts of the code, including the code that draws the preview.

Farhankhosravi, 2021-04-26:
> I am also thinking about this option, because for many it creates problems (when the machine is badly configured or has mechanical defects, in this mode they show up a lot). The purpose, however, is [...]

Thanks. Waiting for that 👍🏻

Farhankhosravi, 2021-04-29:
-UPDATE-
Another bug that i found in constant power mode... Is that sometimes while "cooling" cycle starts, laser will stay on during "cooling" progress. This happened 6 times in 15 cooling cycles... And made my laser very very warmer...
(I'm don't sure that this bug relates to constant power mode, because I have never tested Dynamic mode)

gmmanonymus111, 2021-04-30:
This is most likely not a LaserGRBL issue. Since pwm support is disabled, grbl is set to `lasermode=off`, so when the machine pauses, spindle stays on (so, you will have your laser powered on). When `lasermode` is turned on, grbl will turn the spindle off when your machine pauses.

Farhankhosravi, 2021-04-30:
> This is most likely not a LaserGRBL issue. Since pwm support is disabled, grbl is set to `lasermode=off`, so when the machine pauses, spindle stays on (so, you will have your laser powered on). When `lasermode` is turned on, grbl will turn the spindle off when your machine pauses.

That's wrong, my friend.
`lasermode` is on in both "pwm on" and "off" modes.

gmmanonymus111, 2021-04-30:
Well, for me it turns off the laser every time, wether `Support PWM` is enabled or not as long as `lasermode` is on. Then, my second guess is a hardware issue. (But again, you said sometimes it turns off, so I'm out of ideas)

arkypita, 2021-04-30:
> That's wrong, my friend.
> lasermode is on in both "pwm on" and "off" modes.

Does not matter.

LaserGRBL has no responsibility for this. It just sends a hold / resume exactly as if you pressed the lower right button with your red hand. It is the responsibility of the firmware to turn off the laser during a feed hold, and turn it back on to the previous power at the time of resume.

If you have the problem you describe, it's the firmware's fault, not LaserGRBL.

closed 2021-06-21 by arkypita

arkypita, 2021-06-21:
Fixed this bug in v4.4.0
Currently available as pre-release here https://github.com/arkypita/LaserGRBL/releases/tag/v4.4.0

commit 40375eb1 (2021-06-21): Fix #1434