arkypita/LaserGRBL #1438: when i upgrade it crashes my computer and i have to do a restore to restart computer from ver 4.1 to 4.3

frisbeegolfer69, 2021-04-27:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware and software configuration (please complete the following information):**
 - Windows version
 - Engraver brand and model
 - Grbl version (i.e. 0.9j, 1.1f)
 - LaserGRBL version

**Additional context**
Add any other context about the problem here.

frisbeegolfer69, 2021-04-27:
window 10 
CNC3-3018PRO
laserGRBL VER 4.1.0

arkypita, 2021-04-28:
There is nothing particular between version 4.1 and version 4.3 to justify such behavior.
I think the problem is external.

closed 2021-06-21 by arkypita